bnjmnp/pysoem #1: error when running setup.py

ziadchelala, 2019-07-26:
Hello @bnjmnp, I cloned pysoem into my directory and cloned the soem files into the soem folder inside pysoem.
I ran python3 setup.py build_ext --inplace but I got an error:
error: command 'i686-linux-gnu-gcc' failed with exit status 1.

Below is the message I get after running the setup.py command.

`running build_ext
building 'pysoem.pysoem' extension
i686-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I./pysoem -I./soem/oshw/linux -I./soem/osal/linux -I./soem/oshw -I./soem/osal -I./soem/soem -I/usr/include/python3.5m -c pysoem/pysoem.c -o build/temp.linux-i686-3.5/pysoem/pysoem.o
In file included from ./soem/soem/ethercat.h:18:0,
                 from pysoem/pysoem.c:633:
./soem/soem/ethercatdc.h:20:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
 boolean ec_configdc();
 ^
In file included from /usr/include/python3.5m/pytime.h:6:0,
                 from /usr/include/python3.5m/Python.h:65,
                 from pysoem/pysoem.c:40:
pysoem/pysoem.c: In function ‘__pyx_pf_6pysoem_6pysoem_9CdefSlave_28_get_PO2SOconfig’:
pysoem/pysoem.c:8440:52: error: ‘ec_slavet {aka struct ec_slave}’ has no member named ‘user’
   __Pyx_INCREF(((PyObject *)__pyx_v_self->_ec_slave->user));
                                                    ^
/usr/include/python3.5m/object.h:779:19: note: in definition of macro ‘Py_INCREF’
     ((PyObject *)(op))->ob_refcnt++)
                   ^
pysoem/pysoem.c:8440:3: note: in expansion of macro ‘__Pyx_INCREF’
   __Pyx_INCREF(((PyObject *)__pyx_v_self->_ec_slave->user));
   ^
pysoem/pysoem.c:8441:49: error: ‘ec_slavet {aka struct ec_slave}’ has no member named ‘user’
   __pyx_r = ((PyObject *)__pyx_v_self->_ec_slave->user);
                                                 ^
pysoem/pysoem.c: In function ‘__pyx_pf_6pysoem_6pysoem_9CdefSlave_30_set_PO2SOconfig’:
pysoem/pysoem.c:8507:26: error: ‘ec_slavet {aka struct ec_slave}’ has no member named ‘user’
   __pyx_v_self->_ec_slave->user = ((void *)__pyx_v_self->_cd);
                          ^
pysoem/pysoem.c:8547:42: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
     __pyx_v_self->_ec_slave->PO2SOconfig = __pyx_f_6pysoem_6pysoem__xPO2SOconfi
                                          ^
error: command 'i686-linux-gnu-gcc' failed with exit status 1
`

Please advise on how to proceed.
Thank you.

bnjmnp, 2019-07-28:
Hi @ziadchelala, I guess you checked out  an incompatible SOEM revision. When cloning pysoem, add option `--recurse-submodules` and the compatible SOEM revision will be checked out automatically. Like this:
```
git clone --recurse-submodules https://github.com/bnjmnp/pysoem.git
```
The patches for pysoem to compile are on branch `p2o_user_data` of my SOEM fork.
Provided you cloned my SOEM fork (https://github.com/bnjmnp/SOEM), it should be possible to run `git checkout p2o_user_data` within your soem folder inside pysoem, in order to build pysoem.

I hope this helps.

ziadchelala, 2019-07-29:
This worked, Thank you!

closed 2019-07-29 by ziadchelala