bnjmnp/pysoem #3: Windows 10 download issue

slkirk7, 2019-09-20:
Hello @bnjmnp! 
I have a Beckhoff EK1100 module that I would love to use the PySOEM module for. I am experiencing issues installing the module via pip. I am receiving the error: "LINK : fatal error LNK1181: cannot open input file 'wpcap.lib' ". I believe this is because I don't have WinPcap downloaded, as it is not supported on Windows 10. I tried Npcap in WinPcap compatibility mode, but this did not fix the issue. Do you have any suggestions? Thank you for your help!

bnjmnp, 2019-09-22:
Hi @slkirk7,
I recently installed WinPcap successfully on a Windows 10 system using the official installer from [www.winpcap.org/install](https://www.winpcap.org/install/default.htm). And with that, installing PySOEM was also possible.

On the WinPcap site that is linked above, you can read that the WinPcap is no longer maintained. So far I don't know what it means to switch to Npcap. I guess the switch relies more on contributions to SOEM than to PySOEM.

DK999, 2020-02-05:
Thought WinPcap is kind of deprecated and not maintained in years?

bnjmnp, 2020-10-13:
Using Npcap in WinPcap API-compatible Mode works pretty well.

closed 2020-10-13 by bnjmnp