bnjmnp/pysoem #6: Slave Setup

lemmi25, 2019-11-02:
Hi,

I am new to EtherCat. I got a MXU4300 from Infineon and would like to set up a simple slave master communication. My problem is that I do not understand how to implement the slave in a setup function.

I followed the instruction for the XMC4300 board [here](https://www.infineon.com/dgdl/Infineon-XMC4300_Relax_EtherCat_APP_Slave_SSC-GS-v01_03-EN.pdf?fileId=5546d46254e133b401554f4951cc6447).

The mapping for receive and transmit message can be found here.

[image]

I had a look at the original SOEM as well as the PYSOEM examples (but could not find enough information). Any help on how to setup the setup funtion would be very helpful and mybe a good example for this repo because I think that the XMC4300 is a nice and cheap board.

Many thanks

bnjmnp, 2019-11-05:
Hi lemmi25,

I think your example slave implementation uses fixed PDO mapping so you don't need to do anything in the setup function of your device. Thus you don't even need one.

If you only have the XMC board at you master system something like the following should work:

```py
import pysoem

master = pysoem.Master()

master.open('<put in you network id here!>')

master.config_init()

assert len(master.slaves) == 1

assert master.slaves[0].name == 'XMC_ESC'

master.config_map()

assert master.state_check(pysoem.SAFEOP_STATE, 50000) == pysoem.SAFEOP_STATE

for _ in range(1000):
    print(master.slaves[0].input)

master.state = pysoem.INIT_STATE
master.write_state()

master.close()
```
(Pleas don't use thees asserts in production code, it's just for the purpose of this demonstration)

You should see some bit changes in the printed input values if you push the button on the XMC board, like described in the Infineon PDF on page 56.

If you got there you may easily figure out how to wire outputs to turn on end off the LED on the board. Therefore take a look at the minimal_example.py or basic_example.py, as my example above dose not show how to bring the slave into operational (OP) state.

lemmi25, 2019-11-05:
Hi,

many thanks for the detailed answer. I still have one problem if I use your example I see nothing as blank lines. Even if I push a button. 

If I use the ```minimal_example.py``` I can not get the Slave into operation mode (can send code later, maybe your first answer will resolve my problem). 

Many thanks again.

bnjmnp, 2019-11-11:
Did you find out anything new?
Maybe (I was wrong and) the XMC example uses selectable PDO mapping instead of fixed PDO mapping. If so you need to write to object 0x1C12 and 0x1C13 before calling `config_map()`.
How are objects 0x1C12 and 0x1C13 define?