bnjmnp/pysoem #10: Outputs don't stay set

ngblume, 2020-03-16:
Hej hej,

I managed to get my system to work with pySOEM.
Setup:
- EK1100
- EL3144
- EL2624
- EL2872
- EL 1872

Before starting on the analog stuff, I tried to make the digital input and outputs work.
Got as far as here:

'''
"""Read from modules"""

import sys
import time
import struct
import pysoem

SDO_Info_Check = False

def read_values(ifname):

    # Create EtherCAT master instance
    master = pysoem.Master()
    
    # Open EtherCAT master instance
    master.open(ifname)
    print("EtherCAT master created and started...")

    print("Enumarating slaves")
    print('===========================================================================================')
    # Enumerate and init all slaves
    if master.config_init() > 0:

        # Read state of all slaves at start-up
        master.read_state()

        # Iterate over all slves found
        for slave in master.slaves:
            # Print info on slave
            print('{}:'.format(slave.name))

            # Read state of slave
            print('\tState: {}'.format(hex(slave.state)))

            if (SDO_Info_Check):
                # Check if SDO info is available
                try:
                    od = slave.od
                except pysoem.SdoInfoError:
                    print('\tno SDO info')
                else:
                    for obj in od:
                        print(' Idx: {}; Code: {}; Type: {}; BitSize: {}; Access: {}; Name: "{}"'.format(
                            hex(obj.index),
                            obj.object_code,
                            obj.data_type,
                            obj.bit_length,
                            hex(obj.obj_access),
                            obj.name))
                        for i, entry in enumerate(obj.entries):
                            if entry.data_type > 0 and entry.bit_length > 0:
                                print('  Subindex {}; Type: {}; BitSize: {}; Access: {} Name: "{}"'.format(
                                    i,
                                    entry.data_type,
                                    entry.bit_length,
                                    hex(entry.obj_access),
[... 156 lines cut ...]
        print('no slave available')

    master.close()

if __name__ == '__main__':

    print('script started')

    if len(sys.argv) > 1:
        read_values(sys.argv[1])
    else:
        print('give ifname as script argument')
'''

Problem: 
When I set the outputs, they are set as I want them to be, but only for a split second.
I get the feeling, this is related to DC stuff and maybe me not sending / receiving regularly...
Could you maybe comment if that is the issue or not?

Is sending / receiving regularly required?

Cheers
Niels

P.S.: I'm getting started with EtherCAT in python, and it seems there is alittel bit of context still missing in my understanding of it..

ngblume, 2020-03-18:
Changd my example to have a separate thread taking care of the send and rec'v: works like a charm !!

You can find the example here: https://github.com/ngblume/ethercat-pysoem/tree/master/separate_thread

closed 2020-03-18 by ngblume

bukhari-m, 2024-06-18:
Hi @ngblume this is a very nice example i am trying on my setup too EK1100, EL1008 and EL2008

I see the same error as yours before going on to the example with threading i want to understand, if you can please elaborate for me, that why it blink? What is the underlying reason for this 
Secondly why was there a need to go for a solution with threading?

Please consider that i am new to EtherCAT and new Pysoem.

Thank you so much.