bnjmnp/pysoem #11: Timing of Send_Processdata and Receive_Processdata

ngblume, 2020-03-18:
Another issue / question regarding understanding of timing:

How does the timing between iterations of Send_Processdata and Receive_Processdata work?

In my code (https://github.com/ngblume/ethercat-pysoem/tree/master/separate_thread) I see stable outputs, when running a separate task every 0.01 s (as per your example).
When changing that to every 0.1 s the ouputs work correctly (set and stay), while the input is always read as FALSE.

I still get the feeling, that these some basic EtherCAT timing thing, that I simply haven't understood.
Could you point me in the right direction ? 
Thanks !

Cheers
Niels

bnjmnp, 2020-03-19:
Hi @ngblume,

You are talking about the digital input and digital output devices (EL1872/EL2872)? They don't support DC-Synchronisation, so for both of those guys you don't need any DC settings.

If I understand you right the outputs work fine on both 10ms and 100ms process data cycle time. But the inputs stay FALSE in both cases. So the digital inputs don't work at all?

ngblume, 2020-03-19:
Hi @bnjmnp,

thanks for your answer!
I got it working properly, but I think it would help to get an idea of what DC is and how it is used (when not usind TwinCat)..

Same for the timing in general..
I figured that the processdata thread updates info in the master object with each execution of the send and rec'v cycle. 
So setting an output and directly afterwards reading it back with another module (i.e. setting with EL4114 and reading back with EL3144) might result in no immediate update on the outputs/inputs yet because the processdata thread hasn't run since the changes. Makes sense...

But I might need a little more infos....
Could you point me in the direction of some more info regarding these continuous tasks ?
I'm trying to understand the basic working principles and from there work my way towards DC and setting custom PDOs ...

Thanks !

Cheers
Niels

bnjmnp, 2020-03-22:
Hi Niels,

So far I don't know of any good books about this topic. But I find the datasheet of Beckhoff's ESC IP Core very helpful, and this is freely available on the net:
https://download.beckhoff.com/download/Document/io/ethercat-development-products/ethercat_esc_datasheet_sec1_technology_2i3.pdf 

On the ETG Website you might also find some basic information, ether under [technology](https://www.ethercat.org/en/technology.html) or [here under downloads](https://www.ethercat.org/en/downloads.html) in form of PDF documents.

Good luck!

Benjamin

ngblume, 2020-03-23:
Hi,

thanks.. I will start there reading into the topic..

Last question: Can you give a rough guidance, as to how to set the cycle time for the separate send and receive process data thread? Just in general, not specific, since it clearly relies on the specific system...
Is usually 0.01 sufficient oder even already quite a high load for bus and / or computer / network interface ?

Thanks !

Cheers
Niels

bnjmnp, 2020-03-25:
I'm not an expert but I think 1ms is a usual cycle time for PLCs. But that is dedicated hardware. However on a PC based system 10ms should not be to much of an issue.