bnjmnp/pysoem #12: Add config_overlap_map support

zchen24, 2020-03-23:
This is necessary to support TI based slave devices.

bnjmnp, 2020-03-25:
Hi @zchen24,
This sounds interesting. Could you please give slightly more details!

zchen24, 2020-03-25:
Sure. This is actually is due to a defect in TI's ESC implementation. This is documented in the [SOEM tutorial](Link: https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html) Legacy versus overlapping IOmap section. 

"Overlapping IOmap was introduced to handle the TI ESC that doesn't support RW access to non-interleaved input and output process data of multiple slaves."

bnjmnp, 2020-11-08:
This should be done now, with the 1.0.0 release. There are two new functions for the overlap map mode:

* `config_overlap_map()` that is used instead of the normal `config_map()` function. And
* `send_overlap_processdata()` that is used instead of the normal `send_processdata()`.

Documentation or an example are missing for now.