bnjmnp/pysoem #13: Buffer Overflow error

plipson99, 2020-04-06:
Using both the basic and minimal example scripts, when I try to initialize and send data, I am getting a Buffer Overflow error-whether the cable is plugged in or not. Using Ubuntu v18, python 3.8. I am new to Python so I am wondering how should I reference the ECAT master network ID and are there any special settings I need before using the standard ethernet port as an EtherCAT port? I get a connection error if I enter anything other then the MAC ID of the adapter in the command line as a string, but that is when I get the buffer overflow error.

bnjmnp, 2020-04-13:
Hi @plipson99,
Sorry for the late response.
Please try to run the `find_adapters.py` example or simply run `ifconfig` to get the network ID.
On my Ubuntu machine the correct network ID is `enp3s0`. Any standard ethernet port should work out of the box.

plipson99, 2020-04-20:
Now I am getting a slave not found whether connected or not. Are there settings for the port that should be set, like disabling DHCP and setting a static IP?

Best regards,

Phillip Lipson

plipson99, 2020-04-21:
Nevermind, I figured out my issue. I had 9 adapters and was using the wrong one. Have other issues, but I am communicating with the slave. Thank you.

closed 2020-04-21 by plipson99