bnjmnp/pysoem #14: pysoem on windows 10 problem - will appreciate help

NoamGraetz2, 2020-05-26:
i am trying to use pysoem to emulate a master ethercat controller, but keep banging into a wall.

when i am trying to do pysoem.Master().open(name)   i keep getting
"interface {24D8A70B-00A0-4643-81DE-C1D2DF72396D} could not open with pcap"
message, no matter which interface i am trying to use

this is for all the interfaces detected by pysoem.find_adapters()

also, i have installed win10pcap and wireshark, and they can access any interface 

will appreciate any help

thanks

Noam

NoamGraetz2, 2020-05-26:
well, after some deep diving, i think i have found the problem.
you should use npcap and not win10pcap...

Vimes01, 2021-07-05:
I have the same problem (Windows 10, could not open with pcap, pysoem.find_adapters() finds ID, Wireshark works and can access interface), although I have installed npcap 1.50. Any suggestions?

Vimes01, 2021-07-05:
In my case I needet to input ID in this format: '\\Device\\NPF_{A36B0DE6-FA8F-4113-AC4F-EEFF9E02CEC3}'

msevnctkn, 2022-01-20:
have you solved the problem?

agwidarsito, 2022-10-18:
Weird, on Win 11 we had to do this:

'\Device\\NPF_{A36B0DE6-FA8F-4113-AC4F-EEFF9E02CEC3}'

tomhira, 2023-08-02:
Hello,
I have same problem except I am using Realtek USB GbE Family Controller.
I am using the notation: \Device\NPF_{7A0F6319-EC6D-4EF5-ADC5-52F691B8F63D}
with Winpcap installed.

could not open interface \Device\NPF_{7A0F6319-EC6D-4EF5-ADC5-52F691B8F63D}
  File "C:\basic_example.py", line 426, in run
    self._master.open(self._ifname)
  File "C:\basic_example.py", line 564, in <module>
    BasicExample('\\Device\\NPF_{7A0F6319-EC6D-4EF5-ADC5-52F691B8F63D}').run()
ConnectionError: could not open interface \Device\NPF_{7A0F6319-EC6D-4EF5-ADC5-52F691B8F63D}

What can I do to solve this issue?

tomhira, 2023-08-08:
Hello, it seems the ncap is not properly installed with Windows 10 by selecting the option
Install Npcap in WinPcap API-compatible Mode (WinPcap will be uninstalled)

closed 2024-01-21 by bnjmnp

BorisKomarov, 2024-07-31:
Also you can try this 
master.open(r'\Device\NPF_{B520D7ED-4AC0-48F9-9574-09C0696C3DA4}')
my mistake is that I didn't put "r" in front of the quotes with the interface name to read the slashes correctly