bnjmnp/pysoem #16: How can I add a script for the input slave using basic_example.py?

i1212-blip, 2020-07-10:
EL1008:  input slave
EL2809: output slave 

Below is the following script : 

import sys
import struct
import time
import threading
from collections import namedtuple

import pysoem

class BasicExample:

	BECKHOFF_VENDOR_ID = 0x0002
	EK1100_PRODUCT_CODE = 0x044c2c52
	EL2809_PRODUCT_CODE = 0x0AF93052
	EL1008_PRODUCT_CODE = 0x03F03052
	def __init__(self, ifname):
		self._ifname = ifname
		self._pd_thread_stop_event = threading.Event()
		self._ch_thread_stop_event = threading.Event()
		self._actual_wkc = 0
		self._master = pysoem.Master()
		self._master.in_op = False
		self._master.do_check_state = False

		SlaveSet = namedtuple('SlaveSet', 'name product_code config_func')
		self._expected_slave_layout = {0: SlaveSet('EK1100', self.EK1100_PRODUCT_CODE, None),
							          1: SlaveSet('EL2809', self.EL2809_PRODUCT_CODE, None),
							          2: SlaveSet('EL1008', self.EL1008_PRODUCT_CODE, None)
							           }

	
	def _processdata_thread(self):
		while not self._pd_thread_stop_event.is_set():
			self._master.send_processdata()
			self._actual_wkc = self._master.receive_processdata(10000)
			if not self._actual_wkc == self._master.expected_wkc:
				print('incorrect wkc')
			time.sleep(0.01)

	def _pdo_update_loop(self):
		self._master.in_op = True
		output_len = len(self._master.slaves[1].output)

		print(output_len)
		tmp = bytearray([0 for i in range(output_len)])
		print(tmp)
		toggle = True
		try:
			while 1:
				tmp[0] = 0x01
				tmp[1] = 0x10
				self._master.slaves[1].output = bytes(tmp)
				time.sleep(1)
		except KeyboardInterrupt:
			print('stopped')

	def run(self):
		self._master.open(self._ifname)

		if not self._master.config_init() > 0:
			self._master.close()
			raise BasicExampleError('no slave found')

		for i, slave in enumerate(self._master.slaves):
			if not ((slave.man == self.BECKHOFF_VENDOR_ID) and
					(slave.id == self._expected_slave_layout[i].product_code)):
				self._master.close()
				raise BasicExampleError('unexpected slave layout')
			slave.config_func = self._expected_slave_layout[i].config_func
			slave.is_lost = False

		self._master.config_map()
		self._master.config_dc()

		if self._master.state_check(pysoem.SAFEOP_STATE, 50000) != pysoem.SAFEOP_STATE:
			self._master.close()
			raise BasicExampleError('not all slaves reached SAFEOP state')

		self._master.state = pysoem.OP_STATE

		check_thread = threading.Thread(target=self._check_thread)
		check_thread.start()
		proc_thread = threading.Thread(target=self._processdata_thread)
		proc_thread.start()

		self._master.write_state()

		all_slaves_reached_op_state = False
		for i in range(40):
			self._master.state_check(pysoem.OP_STATE, 50000)
			if self._master.state == pysoem.OP_STATE:
				all_slaves_reached_op_state = True
				break
			#else:
			#    self._master.read_state()
			#    for slave in self._master.slaves:
			#        if not slave.state == pysoem.OP_STATE:
			#            print('{} did not reach OP state'.format(slave.name()))
			#            print('al status code {} ({})'.format(hex(slave.al_status),pysoem.al_status_code_to_string(slave.al_status)))

		if all_slaves_reached_op_state:
			self._pdo_update_loop()
			#self._pdo_update_loop1()

		self._pd_thread_stop_event.set()
		self._ch_thread_stop_event.set()
		proc_thread.join()
		check_thread.join()
		self._master.state = pysoem.INIT_STATE
		# request INIT state for all slaves
		self._master.write_state()
		self._master.close()

		if not all_slaves_reached_op_state:
			raise BasicExampleError('not all slaves reached OP state')

	@staticmethod
	def _check_slave(slave, pos):
		if slave.state == (pysoem.SAFEOP_STATE + pysoem.STATE_ERROR):
			print(
				'ERROR : slave {} is in SAFE_OP + ERROR, attempting ack.'.format(pos))
			slave.state = pysoem.SAFEOP_STATE + pysoem.STATE_ACK
			slave.write_state()
		elif slave.state == pysoem.SAFEOP_STATE:
			print(
				'WARNING : slave {} is in SAFE_OP, try change to OPERATIONAL.'.format(pos))
			slave.state = pysoem.OP_STATE
			slave.write_state()
		elif slave.state > pysoem.NONE_STATE:
			if slave.reconfig():
				slave.is_lost = False
				print('MESSAGE : slave {} reconfigured'.format(pos))
		elif not slave.is_lost:
			slave.state_check(pysoem.OP_STATE)
			if slave.state == pysoem.NONE_STATE:
				slave.is_lost = True
				print('ERROR : slave {} lost'.format(pos))
		if slave.is_lost:
			if slave.state == pysoem.NONE_STATE:
				if slave.recover():
					slave.is_lost = False
					print(
						'MESSAGE : slave {} recovered'.format(pos))
			else:
				slave.is_lost = False
				print('MESSAGE : slave {} found'.format(pos))

	def _check_thread(self):

		while not self._ch_thread_stop_event.is_set():
			if self._master.in_op and ((self._actual_wkc < self._master.expected_wkc) or self._master.do_check_state):
				self._master.do_check_state = False
				self._master.read_state()
				for i, slave in enumerate(self._master.slaves):
					if slave.state != pysoem.OP_STATE:
						self._master.do_check_state = True
						BasicExample._check_slave(slave, i)
				if not self._master.do_check_state:
					print('OK : all slaves resumed OPERATIONAL.')
			time.sleep(0.01)

class BasicExampleError(Exception):
	def __init__(self, message):
		super(BasicExampleError, self).__init__(message)
		self.message = message

if __name__ == '__main__':

	print('basic_example started')

	if len(sys.argv) > 1:
		try:
			BasicExample(sys.argv[1]).run()
		except BasicExampleError as expt:
			print('basic_example failed: ' + expt.message)
			sys.exit(1)
	else:
		print('usage: basic_example ifname')
		sys.exit(1)

bnjmnp, 2020-07-13:
Hi @i1212-blip,

It is difficult to help. Where do you stuck?

Maybe it is better for you to start from the `minimal_example.py`. And in the first steps you may use only one Beckhoff terminal.

If you are beginning with EtherCAT right away, it might be helpful to play with TwinCAT a little bit first.

Site note: If you place triple backticks \`\`\` before and after code blocks, within your GitHub comments, you get nicer code blocks. You can even have syntax highlighting: [https://help.github.com/en/articles/creating-and-highlighting-code-blocks](https://help.github.com/en/articles/creating-and-highlighting-code-blocks).

i1212-blip, 2020-07-14:
Hello,
I'm running my new script using two motors EL7201, and I'm getting the following error: 

basic_example started
incorrect wkc
EL7201 did not reach OP state
al status code 0x0 (No error)
basic_example failed: not all slaves reached OP state

Below is the new script: 

```

import sys
import struct
import time
import threading
from collections import namedtuple

import pysoem

class BasicExample:

    BECKHOFF_VENDOR_ID = 0x0002
    EK1100_PRODUCT_CODE = 0x044c2c52
    EL2809_PRODUCT_CODE = 0x0AF93052
    EL1008_PRODUCT_CODE = 0x03F03052
    EL7201_PRODUCT_CODE = 0x1C213052

    def __init__(self, ifname):
        self._ifname = ifname
        self._pd_thread_stop_event = threading.Event()
        self._ch_thread_stop_event = threading.Event()
        self._actual_wkc = 0
        self._master = pysoem.Master()
        self._master.in_op = False
        self._master.do_check_state = False

        SlaveSet = namedtuple('SlaveSet', 'name product_code config_func')
        self._expected_slave_layout = {0: SlaveSet('EK1100', self.EK1100_PRODUCT_CODE, None),
                                       # 1: SlaveSet('EL2809', self.EL2809_PRODUCT_CODE, self.el2809_setup)
                                       1: SlaveSet('EL2809', self.EL2809_PRODUCT_CODE, None),
                                       2: SlaveSet('EL1008', self.EL1008_PRODUCT_CODE, None),
		                       3: SlaveSet('EL7201', self.EL7201_PRODUCT_CODE, self.el7201_setup1),
				       4: SlaveSet('El7201', self.EL7201_PRODUCT_CODE, self.el7201_setup2)
				        }

    def el7201_setup1(self, slave_pos):
         slave = self._master.slaves[slave_pos]
         map_1c12_bytes = struct.pack('BxHH', 2, 0x1600, 0x1601)
         slave.sdo_write(0x1c12, 0, map_1c12_bytes, True)
         slave.dc_sync(1, 1000000, 0, 1000000)

    def el7201_setup2(self, slave_pos):
         slave = self._master.slaves[slave_pos]
         map_1c12_bytes = struct.pack('BxHH', 2, 0x1600, 0x1601)
         slave.sdo_write(0x1c12, 0, map_1c12_bytes, True)
         slave.dc_sync(1, 1000000, 0, 1000000)
   

    def _processdata_thread(self):
        while not self._pd_thread_stop_event.is_set():
            self._master.send_processdata()
            self._actual_wkc = self._master.receive_processdata(10000)
            if not self._actual_wkc == self._master.expected_wkc:
                print('incorrect wkc')
            time.sleep(0.01)

     def _pdo_update_loop(self):
	
[... 119 lines cut ...]
                        self._master.do_check_state = True
                        BasicExample._check_slave(slave, i)
                if not self._master.do_check_state:
                    print('OK : all slaves resumed OPERATIONAL.')
            time.sleep(0.01)

class BasicExampleError(Exception):
    def __init__(self, message):
        super(BasicExampleError, self).__init__(message)
        self.message = message

if __name__ == '__main__':

    print('basic_example started')

    if len(sys.argv) > 1:
        try:
            BasicExample(sys.argv[1]).run()
        except BasicExampleError as expt:
            print('basic_example failed: ' + expt.message)
            sys.exit(1)
    else:
        print('usage: basic_example ifname')
        sys.exit(1)
```

bnjmnp, 2020-07-14:
Hi @i1212-blip,

I think once you solve the "did not reach OP state" issue, the incorrect wkc notification might go away. If you only get the "incorrect wkc" only once during startup this should not be too much of an issue.

So first you should focus on getting the EL7201 into OP state. Strangely the "al status code" is not very helpful. I think the problem why the EL7201 is not going into OP stat is that there are some settings missing. Taking a look at the manuals section "6.2.3 Settings in the CoE register" ([https://download.beckhoff.com/download/document/io/ethercat-terminals/el72x1en.pdf](https://download.beckhoff.com/download/document/io/ethercat-terminals/el72x1en.pdf)), there are a lot of settings applied during transition from PreOP to SafeOP state, maybe thees are checked when going to OP state. So what you see there in the "Startup-Tab" must go into the el7201_setup1() function.

The EL7201 looks pretty komplex, ..hope you get this working.

i1212-blip, 2020-07-15:
I updated my el7201_setup1 like Startup-Tab : 
   ``` def el7201_setup1(self, slave_pos):
         slave = self._master.slaves[slave_pos]
         tx_map_1c13_bytes = struct.pack('BxHH', 2, 0x1A00, 0x1A01)
         slave.sdo_write(0x1c13, 0, tx_map_1c13_bytes, True)
         rx_map_1c12 = [0x1600, 0x1601]
         rx_map_1c12_bytes = struct.pack('Bx' + ''.join(['H' for i in range(len(rx_map_1c12))]), len(rx_map_1c12),
                                        *rx_map_1c12)

         slave.sdo_write(0x1c12, 0, rx_map_1c12_bytes, True)
         slave.dc_sync(1, 10000000, 1000000, 0)
```
and I'm still getting the same error EL7201 did not reach OP state. Any suggestions?

bnjmnp, 2020-07-15:
You write only 2 two bytes to the 4 byte object entries.
In other words, for the uint32 objects entries you need another format character in you `struct.pack` call
So for example you need to do

```py
u32val7 = 150
slave.sdo_write(0x8010, 0x14, struct.pack("I", u32val7), True)
```

instead of:

```py
u32val7 = 150
slave.sdo_write(0x8010, 0x14, struct.pack("H", u32val7), True)
```
See the format chracters for each type [https://docs.python.org/3/library/struct.html#format-characters](https://docs.python.org/3/library/struct.html#format-characters)

Something like this might even be more readable:

```py
slave.sdo_write(0x8010, 0x14, (150).to_bytes(4, byteorder='little') , True)
```
See the [`int.to_bytes`](https://docs.python.org/3/library/stdtypes.html#int.to_bytes) function.

For the uint16 objects entries it must be `(<nbr>).to_bytes(2, byteorder='little')`.

bnjmnp, 2020-07-15:
Ahja, and for the `#Motor pole pairs (0x4)` it must be:

```py
u8val = 4
slave.sdo_write(0x8011, 0x13, struct.pack("B",u8val), True)
```

or

```py
slave.sdo_write(0x8011, 0x13, (4).to_bytes(1, byteorder='little'), True)
```

bnjmnp, 2020-07-22:
Hi @i1212-blip, did you had any success? I would appreciate some feedback, at least an info if this can be closed.

bnjmnp, 2021-02-28:
Closed due to inactivity of the original author.

closed 2021-02-28 by bnjmnp