bnjmnp/pysoem #17: Problem installing PySOEM on Win7

marcAC, 2020-07-30:
Hey,

I'm trying to install PySOEM on my Win 7 machine but I get the following error:

LINK : fatal error LNK1181: Eingabedatei "wpcap.lib" kann nicht geöffnet werden.
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.14.26428\\bin\\HostX86\\x86\\link.exe' failed with exit status 1181

Currently I have the newest version of Npcap installed, but before I tried the latest version of Winpcap with the same result.
I installed the package Cython and I have Visual Studio 2017 installed,

Do you have anything in mind how to solve this problem?

Thanks in advance and kind regards,
Marc

bnjmnp, 2020-08-01:
Hi @marcAC,
I guess you are using a 32-Bit Python interpreter. This won't work on windows. You better use 64-Bit Python. The PySOEM package contains a binary wheel for this, thus you then need no compiler.

marcAC, 2020-08-04:
Hey @bnjmnp,

I indeed was using a 32-Bit Python interpreter, with a 64-bit interpreter the installation process went perfectly fine. Thank you!

closed 2020-08-04 by marcAC