bnjmnp/pysoem #20: wrap the `ec_send_overlap_processdata`

jackie-wayland, 2020-09-14:


bnjmnp, 2020-09-14:
Hi @jackie-wayland,

thanks for the suggestion. I think this a duplicate to #12.
I'm not very into the overlap topic. I can implement the wrapper code for thees functions straight away, but I need someone to test it. Do you have a good hardware setup to test the overlapping IOmap? So once I implemented this could you test it?

jackie-wayland, 2020-09-22:
@bnjmnp I have a hardware setup, and I can test it.

bnjmnp, 2020-10-11:
@jackie-wayland witch platform do you plan to use for testing?

bnjmnp, 2020-10-18:
Hey @jackie-wayland,

I have build a new version of the pysoem package that adds the overlap stuff. For now I uploaded it to [TestPyPI](https://test.pypi.org/) only. You can install it via `python -m pip install --index-url https://test.pypi.org/simple/ pysoem`. It is version 0.1.2. The changes in the source are on branch [overlappion_iomap](https://github.com/bnjmnp/pysoem/tree/overlapping_iomap). 

With that changes there are two new function for the pysoem.Master class:
* `config_overlap_map()` that is used instead of the normal `config_map()` function. And
* `send_overlap_processdata()` that is used instead of the normal `send_processdata()`.

I hope with that, devices based on TI ESCs can be used with pysoem.

It would be very nice if you could test this!
Many thanks in advance.

jackie-wayland, 2020-10-19:
@bnjmnp  I'll try it on Kollmorgen AKD drive.

bnjmnp, 2020-10-24:
Hi @jackie-wayland,

Do you know if the Kollmorgen AKD drive is based on a TI ESC, or if the drive requires the overlap iomap?
This info might be helpful for the author of issue #21.

jackie-wayland, 2020-10-25:
@bnjmnp  I'm not sure about whether the Kollmorgen AKD drive is based on a TI ESC. But I test this branch, the PDO buffers are arranged as the overlap iomap,  the drive also works as expected.

bnjmnp, 2020-11-08:
Now the above described overlap iomap support is merged into the master branch and also part of the latest pyseom release that can be installed via pip. Just documentation or an example is missing.