bnjmnp/pysoem #21: how to send NOP command frame

jackie-wayland, 2020-09-22:
I'm testing the Kollmorgen AKD2G drive, but the pysoem can't work correctly with it. I captured the frame from PCMM (Kollmorgen EtherCAT master)  to AKD2G drive, and I find that the PCMＭ sends NOP command periodically to keep something. 

Can the pysoem send the NOP command?  How to enable this?

bnjmnp, 2020-09-22:
It is not possible to send the NOP command with pysoem, same with SOEM I guess, if you use the available API functions.

I don't think that the missing NOP command is the problem here. The NOP command should pretty much be ignored by the drive.
Did you try to send and receive the process data cyclically, like in the examples? (using function send_processdata and receive_processdata)
I guess the drive dose not go to operational state, right? Is at least SDO communication working?

jackie-wayland, 2020-09-23:
@bnjmnp the drive goes to OP state, but the DC is not enable. `send_processdata` and `receive_processdata` are used cyclically. But the `dc_sync()` not work, I don't know the alternatives to this function. I always get the `PLL is lost` error from the drive after enable the drive.

jackie-wayland, 2020-09-23:
If the `dc_sync()` not work, the PDO frame will not include the DC information?

bnjmnp, 2020-09-23:
Not sure. Did you try calling `config_dc()` after the call to `config_map()`. This may be required to get de_sync working. How do you see that `dc_sync()` is not working? Did you try changing the rate on witch `send_processdata` and `receive_processdata` are called?

jackie-wayland, 2020-09-27:
@bnjmnp I configured the DC after config_map(), but the `ado 0x981` of the slave did not show the enable sync0 information. Could you wrap the [ethercat base command](https://github.com/OpenEtherCATsociety/SOEM/blob/master/soem/ethercatbase.c), I think this is very convenient for basic debugging.

bnjmnp, 2020-09-27:
Thees base functions are the lower layer of the SOEM stack. They are not meant to be called from the application. In fact wrapping them won't go along well with the current pysoem API.
* Did you try building a C application directly with SOEM?
* Did you try to get in contact with Kollmorgen support on the NOP thing?
* Or did you got the drive running using a Windows PC with TwinCAT?