bnjmnp/pysoem #23: pdo output does not seem to work

ziafon, 2020-10-07:
I wrote a program to drive the motor with pysoem. It reached op state successfully but could only receive the data from the driver. If i try to send the control world to the driver, it doesn't work. Is this sth. wrong? Thanks for your help.

import sys
import pysoem
import struct
import time

class MotorDriver:

    def __init__(self, ifname):
        self._ifname = ifname
        self._actual_wkc = 0
        self._master = pysoem.Master()
        self._master.in_op = False
        self._master.do_check_state = False
    
    def pdo_map(self,pos):
        slave = self._master.slaves[pos]

        slave.sdo_write(0x1C12, 0, struct.pack('B', 0))
        slave.sdo_write(0x1C13, 0, struct.pack('B', 0))

        slave.sdo_write(0x1A02, 0, struct.pack('B', 0))
        slave.sdo_write(0x1A02, 1, struct.pack('I', 0x60410010))
        slave.sdo_write(0x1A02, 2, struct.pack('I', 0x60640020))
        slave.sdo_write(0x1A02, 0, struct.pack('B', 2))   

        slave.sdo_write(0x1602, 0, struct.pack('B', 0))
        slave.sdo_write(0x1602, 1, struct.pack('I', 0x60400010))
        slave.sdo_write(0x1602, 2, struct.pack('I', 0x607A0020))
        slave.sdo_write(0x1602, 0, struct.pack('B', 2))  

        slave.sdo_write(0x1C12, 1, struct.pack('H', 0x1602))
        slave.sdo_write(0x1C12, 0, struct.pack('B', 1))

        slave.sdo_write(0x1C13, 1, struct.pack('H', 0x1A02))
        slave.sdo_write(0x1C13, 0, struct.pack('B', 1)) 

    def motor_run(self):    

        # Open EtherCAT master instance
        self._master.open(self._ifname)
        print("EtherCAT master created and started...")

        print("Enumarating slaves")
        # Enumerate and init all slaves
        if self._master.config_init() > 0:

            print("{} slaves found and configured".format(len(self._master.slaves))
     
            # PDO config 
            self.pdo_map(0)
            self.pdo_map(1)
            
            # PREOP_STATE to SAFEOP_STATE request - each slave's config_func is called
            self._master.config_map()
            print('Slaves mapped, state to SAFE_OP')

            # Read state of all slaves at start-up
            if self._master.state_check(pysoem.SAFEOP_STATE, 50000) != pysoem.SAFEOP_STATE:    
                self._master.read_state()
                for slave in self._master.slaves:
                    if not slave.state == pysoem.SAFEOP_STATE:
                        print('{} did not reach SAFEOP state'.format(slave.name))
                        print('al status code {} ({})'.format(hex(slave.al_status),
                                                              pysoem.al_status_code_to_string(slave.al_status)))
                raise Exception('not all slaves reached SAFEOP state')

            self._master.state = pysoem.OP_STATE
            self._master.write_state()

            self._master.state_check(pysoem.OP_STATE, 50000)
            if self._master.state != pysoem.OP_STATE:
                self._master.read_state()
                for slave in self._master.slaves:
                    if not slave.state == pysoem.OP_STATE:
                        print('{} did not reach OP state'.format(slave.name))
                        print('al status code {} ({})'.format(hex(slave.al_status),
                                                              pysoem.al_status_code_to_string(slave.al_status)))
                raise Exception('not all slaves reached OP state')

            try:

                self._master.slaves[0].output = struct.pack('H',0x06)
                self._master.slaves[1].output = struct.pack('H',0x06)
                self._master.send_processdata()
                time.sleep(0.001)
                self._master.slaves[0].output = struct.pack('H',0x07)
                self._master.slaves[1].output = struct.pack('H',0x07)
                self._master.send_processdata()
                time.sleep(0.001)
                self._master.slaves[0].output = struct.pack('H',0x0F)
                self._master.slaves[1].output = struct.pack('H',0x0F)
                self._master.send_processdata()
                time.sleep(0.001)

                while 1:
                    # free run cycle
                    self._master.send_processdata()
                    self._master.receive_processdata(1000)
                    
                    rec1=self._master.slaves[0].input
                    rec2=self._master.slaves[1].input
                    
                    p=2
                    data1=struct.unpack_from('!i',rec1,p)
                    data2=struct.unpack_from('!i',rec2,p)

                    print('ac_ps:',data1,'    ','ac_ps:',data2)
   
                    time.sleep(0.01)

            except KeyboardInterrupt:
                # ctrl-C abort handling
                print('stopped')
            self._master.state = pysoem.INIT_STATE
            # request INIT state for all slaves
            self._master.write_state()
            self._master.close()
        
        else:
            print('slaves not found')

if __name__ == '__main__':

    print('script started')

    if len(sys.argv) > 1:
        try:
            MotorDriver(sys.argv[1]).motor_run
        except Exception as expt:
            print(expt)
            sys.exit(1)
    else:
        print('give ifname as script argument')
        sys.exit(1)

bnjmnp, 2020-10-09:
Hi @ziafon,

Dose your process data output really only consists of the 2 byte control word?

Maybe your drives only work with DC synchronization enabled, so you need to enable DC with function `dc_sync()` like in the `basic_example.py`.
You also may need to start the process data cycle prior to going to operational state, via a thread that runs in parallel. For this take a look at the `basic_example.py`.

Site note: If you place triple backticks \`\`\` before and after code blocks, within your GitHub comments, you get nicer code blocks. You can even have syntax highlighting as described [here](https://help.github.com/en/articles/creating-and-highlighting-code-blocks). People can read and understand your code better and faster.

bnjmnp, 2021-06-06:
Hi @ziafon,
did you actually found a solution for this?

bnjmnp, 2024-01-21:
Closed due to inactivity of the OP.

closed 2024-01-21 by bnjmnp