bnjmnp/pysoem #25: How to use

GabrielGomesBarbosa, 2020-10-24:
Hi folks, im new in Raspberry and etherchat.

I cloned it on my raspberry and try to run it, but when i run `setup.py` i get a message on terminal.

```
sage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied
```

My goal is use this lib to connect a raspberry with an ethercat slave machine.

Can someone help me, to run it?

bnjmnp, 2020-10-25:
Hi Gabriel,

not sure which OS you are using for the pi (I guess its the Raspberry Pi OS), but if you have internet access you could install pysoem from PyPI via `python3 -m pip install pysoem`. It might be required to install Cython beforehand via `python3 -m pip install Cython`.
If you like to install pysoem from the sources you should know that you need to clone the repo recursively via `git clone --recursive https://github.com/bnjmnp/pysoem.git`. With the sources there you install the package via `python3 setup.py install`.

I recommend to use virtualenv for trying things out:

* `python3 -m pip install virtualenv` installs virtualenv
* with `python3 -m virtualenv venv` you creat a "virtualenv"
* with `source venv/bin/activate` you activate the virtualenv
* you now can use python3 and install packages as it would be your systems Python but you can remove the virtualenv by deleting the venv directory without any harm to the systems Python interpreter. You also have less access violation issues.

GabrielGomesBarbosa, 2020-10-25:
Hi @bnjmnp 

Yes i'm using Raspberry PI OS, thanks for your tips... i will use the `virtualenv`.

If i do this... i will be ale to run the `minimal_example.py` ? when i ran this this example with `python mininal_example.py` i get this message on console: `usage: basic_example ifname`

```
    if len(sys.argv) > 1:
        try:
            BasicExample(sys.argv[1]).run()
        except BasicExampleError as expt:
            print('basic_example failed: ' + expt.message)
            sys.exit(1)
    else:
        print('usage: basic_example ifname')
        sys.exit(1)
```
because i dont send args... my question is:

If i use `virtualenv` will solve it ?
Which argument i need to pass to execute `minimal_example.py` ?

bnjmnp, 2020-10-25:
You need to give the ID of your network interface. Sure your Pi only has one, but other systems may have more. In your case its probably `eth0` so you do `python3 minimal_example.py eth0`.
You get a list of available interfaces by calling the `ifconfig` in the terminal. Also pysoem/examples/find_adapters.py should give you the same list.
The virtualenv is not neede for this.

GabrielGomesBarbosa, 2020-10-25:
@bnjmnp thanks for your support, i will test it... and after the test, i will back here to tell the result.

GabrielGomesBarbosa, 2020-10-26:
Hi @bnjmnp i have new question...

Look, i still not have a slave machine to test (but i will have one very soon), so for while i'm testing in my home with standard Ethernet port, without W5500 Ethernet Shield...

I ran the `find_adaters.py` and i get this list:

[image]

After that i choose the eth0, and i get this message:

[image]

My question is:

Shouldn't I get a message saying "No slaves found" ? or this error is normal because i don't have a slave connected ?

bnjmnp, 2020-10-26:
Yes the error "ConnectionError not found" is related to using Python 2, as the built-in exception ConnectionError came first up with Python 3.

I guess the "could not open interface eth0" error is thrown because you need root access to open eth0 for raw ethernet communication. So try to do `sudo python3 minimal_example.py eth0`.

GabrielGomesBarbosa, 2020-10-28:
@bnjmnp Thanks, now its working "Slaves not found".

[image]

closed 2020-10-28 by GabrielGomesBarbosa