bnjmnp/pysoem #26: Documentation and PDO/SDO Maps

GabrielGomesBarbosa, 2020-11-21:
Hi @bnjmnp,

Thanks for help me in #25 its work when i tested the `minimal_example.py` on real ETHERCAT communication, but... I still have some doubts, and i saw that in `examples` folder have `read_sdo_info.py`, it will be the next test.

My question is:

1. How can i write on SDO's ? 
2. How to read and write PDO's ?
3. There is some function that list all maps PDO/SDO index and address in all slaves ?
4. There is some documentation that list all functions in `pysoem` and what it do ? Some functions have parameters that i dont know what is... and it will help me to build without send questions here.

GabrielGomesBarbosa, 2020-11-21:
1. To write SDO i found this example `slave.sdo_write(0x8001, 2, struct.pack('B', 1))` but I don't know how you got there.

bnjmnp, 2020-11-22:
For now, apart from the examples, there is no more documentation. The interactive help is also not to much of a help.

But thanks four your list, I think this lists the things a user may want to know.
I`ll take this into account when creating documentation.

bnjmnp, 2020-11-22:
In the example of sdo_wirte: `slave.sdo_write(0x8001, 2, struct.pack('B', 1))`. The first parameter is the index(0x8001), the second parameter is the subindex(2) and then third parameter is the data to be written to the object. The data parameter is always a `bytes` object.

GabrielGomesBarbosa, 2020-11-22:
Hi @bnjmnp thanks for your answer... can i help build that doc ? i sent messages on Gitter chat, i understood now how to write in SDO.

This library is a Cython wrapper from `EtherCAT Master Library`, and that library we have a function to list PDO Mapping.

[image]

In folder `examples` i guess that `read_sdo_info.py` will show SDO Mapping right ?

There is some function to list all PDO Mapping ? If yes... can you send me here an example ?

bnjmnp, 2020-11-23:
Your screenshoot is the terminal output of the slaveinfo example from the SOEM tests. When you take a closer look to the source code (slaveinfo.c) you will see (within function `si_map_sdo`) that there is not directly a SOEM library function that prints the PDO mappings. They use a bunch of `ec_SDOread`s.
You could do something similar with PySOEMs `sdo_read`. Unfortunately the `read_sdo_info.py` is not printing the PDO mappings.

GabrielGomesBarbosa, 2020-11-23:
@bnjmnp thanks for your answer, i really appreciate.

I'm studying the `sdo_read` function here, but i still got some difficult about it. Can you provide some example about to get PDO map using this function ?

bnjmnp, 2020-11-23:
First you need to known the EtherCAT PDO mapping concept.  Unfortunately that is not very easy to explain, and I cannot do it in every detail.

You already know thees CoE object, you are trying to read with `sdo_read`, I guess. Each object can be addressed by it's index (0 to 0xFFFF) and subindex (0 to 255). These objects are used for settings of a device. For example if you have a temperatur sensor device you could have a unit object witch is use to switch the unit of the output from degree Celsius to Fahrenheit scale.
Within these objects there are a bunch of standard objects, which have special meanings. For example the object with the index 0x1018, that gives a master the identity info of a device. [Here](https://infosys.beckhoff.com/english.php?content=../content/1033/ep4174/3386579467.html&id=) is a nice example of the available standard objects of a Beckoff EP4174.
The PDO mapping can also be read out of the standard objects. Most important are the objects 0x1C12 for output PDO and 0x1C13 for input PDO. And then all objects in the range of 0x1600 to 0x 17FF and 0x1A00 to 0x1B00 are from interest.
Try to read out the content of both objects 0x1C12 and 0x1c13 with `sdo_read`. Or in an earlier step try to read out some of the standard objects in the range of 0x1000 to 0x1018 to get familiar with `sdo_read`.

Take also a look at [my comment in issue #18](https://github.com/bnjmnp/pysoem/issues/18#issuecomment-669442192) where I explain what `sdo_read` returns.