bnjmnp/pysoem #28: SDO accesses interfere PDO update loop?

tanakarei256, 2021-01-15:
I'm writing a motor control program based on basic_example.py.
Sometimes I get a sync manager watchdog error when I make SDO accesses while running a PDO update loop.
I guess this is because the SDO access function interferes with the PDO update loop thread by the GIL.
I think it can be solved by using a non-blocking SDO access function or by separating the PDO update loop and SDO access into two processes.
However, both seem to be difficult with the current pysoem.

Thank you.

bnjmnp, 2021-01-16:
I can't really tell. For now I can't even think on how to do this nateivly with the SOEM API itself. Maybe you could find something about this in the [SOEM issue list](https://github.com/OpenEtherCATsociety/SOEM/issues?q=is%3Aissue).

tanakarei256, 2021-01-18:
Thank you for your reply.

To solve the problem fundamentally, I would like to add a process in my program that performs a PDO update cycle.

I referred to this issue: OpenEtherCATsociety/SOEM/issues/405.
According to this discussion, SOEM is designed to be thread-safe except for PDO.
Therefore, I think this issue can be resolved by adding a process for the PDO update cycle.
Locking other functions will prevent the PDO update cycle from being interrupted.
However, I find it difficult to implement the PDO update cycle process myself.
This is because pysoem.Master is not picklable and cannot be shared by functions in python's multiprocessing module.
Therefore, I think the process of PDO update cycle should be implemented inside pysoem.

If that is not possible, simply making the timeout arguments for sdo_read and sdo_write adjustable may alleviate this problem.
It is currently given an immutable value, EC_TIMEOUTRXM = 700,000 us = 700 ms, which is too long to run at the same time as the PDO update cycle.

bnjmnp, 2021-01-18:
Putting PDO update cycle in a process inside pysoem, sounds good to me. But to figure out how to implement this, I need to take a closer look at things. You still need a way to update the PDO data, synchronized with the PDO process.
Making the SDO timeouts adjustable is much more a short term issue, and is already on my todo list. Here I think of a `Master.sdo_timeout` property that is applied to the SOEM SDO read and write. But maybe this only possible on a per device basis.

zopcuk, 2021-01-19:
Hi @tanakarei256 ,
I have same problem right now.
I read your comments and this issue [https://github.com/OpenEtherCATsociety/SOEM/issues/405](url)
And mostly understand problem but ı dont know how to do it really. Can i solve problem with pysoem ? And can you explain me little bit more ?
Thank you.

tanakarei256, 2021-01-20:
I'm not familiar with SOEM or EtherCAT, so I don't know how to implement it.
For my purposes, I think it's enough to have a function that waits for the next PDO update for synchronization.
However, more features may be needed for other uses.

bnjmnp, 2021-01-26:
Hi @tanakarei256,
at least the SDO timeouts are now configurable. Check out [pysoem.Master.sdo_read_timeout](https://pysoem.readthedocs.io/en/latest/master.html#pysoem.Master.sdo_read_timeout) and [pysoem.Master.sdo_write_timeout](https://pysoem.readthedocs.io/en/latest/master.html#pysoem.Master.sdo_write_timeout). This will be availabel in the latest version of pysoem (1.0.2) installed via pip from PyPI.

tanakarei256, 2021-01-26:
Thanks for the improvement.
I think that shortening the timeout for SDO communication has reduced the problem.

jinkumak, 2021-06-04:
Can I control motor drive ?
Can  I move motor ? 
please give me a sample code

bnjmnp, 2021-06-06:
Hi @jinkumak,
with pysoem you should be able to control any EtherCAT actor device, or retrive data from any EtherCAT sensor device.
You could have a look at issue #30 as a starting point, there is an example for a CiA 402 device profile based motor drive.
Please open a new issue for any further discussion.