bnjmnp/pysoem #29: License question

g3n35i5, 2021-01-15:
Dear @bnjmnp,

thank you very much for this great wrapper! I would like to know why you chose a GPL-2.0 licence, which allows commercial use, but only under the condition that the source code must be published.

Therefore I have two questions:
1. would you consider putting the project under a different licence (e.g. MIT)?
2. if you do not want to do this, would you license the project, i.e. give a company the right to use the code without having to publish its own source code?

Many greetings and many thanks

bnjmnp, 2021-01-16:
Thanks @g3n35i5.

First of all I'm really open to put pysoem under a more permissive licence. But as SOEM itself is under a GPL-2.0 licence, and the wrapper directly compiles SOEM code, I decided to put pysoem also under GPL-2.0. Thus I don't get into trouble. And for the ease of use, it is really good to compile SOEM directly into pysoem, thus one dose not need to compile SOEM separately in order to use pysoem.

However SOEM has a special exception in it's [licence terms](https://github.com/OpenEtherCATsociety/SOEM/blob/master/LICENSE). I'm not good with leagel issues. Somehow it reads as it would be okay to put pysoem under another licence. But I'm not sure. That is also because I use a slightly patched version of SOEM.

g3n35i5, 2021-01-16:
Thanks @bnjmnp for your quick answer!

I totally understand from this aspect why you chose the same license.

To make it as transparent as possible: I want to use pysoem in my company in a commercial project. We already have a license to use SOEM. I will have a closer look at the license terms on monday and get back to you as soon as possible!

Best regards and many thanks!

g3n35i5, 2021-01-19:
Hello @bnjmnp!

I'm sorry that I didn't manage to get back to you until yesterday.
Now, however, I have looked more closely at the licensing issues.

Your fork of SOEM uses the same license as SOEM without modifications. The GPLv2 License allows you to do so as long as you're publishing it. So your fork is safe.

As you've mentioned correctly, the SOEM license (which inherits from the GPL2.0 license) and thus the license of your fork
gives you a special permission so that you don't have to put pysoem under the same license as SOEM:

> As a special exception, if other files instantiate templates or use macros or
> inline functions from this file, or you compile this file and link it with other
> works to produce a work based on this file, this file does not by itself cause
> the resulting work to be covered by the GNU General Public License. However the
> source code for this file must still be made available in accordance with
> section (3) of the GNU General Public License.

So let's have a look at the corresponding section in the GPL2 license:

> 3. You may copy and distribute the Program (or a work based on it, under Section 2) in object code or executable form under the terms of Sections 1 and 2 above provided that you also do one of the following:
> 
> a) Accompany it with the complete corresponding machine-readable source code, which must be distributed under the terms of Sections 1 and 2 above on a medium customarily used for software interchange; or,
> b) Accompany it with a written offer, valid for at least three years, to give any third party, for a charge no more than your cost of physically performing source distribution, a complete machine-readable copy of the corresponding source code, to be distributed under the terms of Sections 1 and 2 above on a medium customarily used for software interchange; or,
> c) Accompany it with the information you received as to the offer to distribute corresponding source code. (This alternative is allowed only for noncommercial distribution and only if you received the program in object code or executable form with such an offer, in accord with Subsection b above.)

So far so good, you've published the pysoem code, so i think you can put pysoem under a more permissive license, e.g. Apache/MIT/BSD.

Another useful comment which underlines my point of view can be found here: https://github.com/OpenEtherCATsociety/SOEM/issues/44.

Disclaimer: I am not a lawyer, just a programmer, and as such none of the above should be construed as legal advice.

bnjmnp, 2021-01-19:
Thank you for your research, this all sounds plausible to me. I'll think through this a one or two more days.

bnjmnp, 2021-01-21:
So, it's done, I made the change.
A new version on PyPI is coming soon.

closed 2021-01-21 by bnjmnp

g3n35i5, 2021-01-22:
Great thank you!

wmyqcy, 2025-03-26:
> 感谢您的快速回答！
> 
> 从这方面我完全理解您为什么选择相同的许可证。
> 
> 尽可能透明：我想在我公司的商业项目中使用 pysoem。我们已经拥有使用 SOEM 的许可证。我将在周一仔细查看许可条款，并尽快回复您！
> 
> 最诚挚的问候和非常感谢！

Do you already have the SOEM license, and have you applied for the EtherCAT master license from Beckhoff Automation GmbH?

wmyqcy, 2025-03-26:
> Thanks [@bnjmnp](https://github.com/bnjmnp) for your quick answer!
>
> I totally understand from this aspect why you chose the same license.
>
> To make it as transparent as possible: I want to use pysoem [...]

Do you already have the SOEM license, and have you applied for the EtherCAT master license from Beckhoff Automation GmbH?