bnjmnp/pysoem #31: PySOES

dreaquil, 2021-02-06:
Are there any plans to begin a PySOES project? If not, any recommendations of where to get started?

bnjmnp, 2021-02-07:
I have no plans to start a PySOES project.

To build an EtherCAT slave, one need more specialized hardware. For example when building a slave that is based on an Raspberry Pi, you need a shield like this: [https://www.sg-electronic-systems.com/ethercat-shield-for-raspberry-pi-etherc-v1-6/](https://www.sg-electronic-systems.com/ethercat-shield-for-raspberry-pi-etherc-v1-6/).

PySOEM has it's difficulties when hard real-time is required, which is the case for most industrial appliances. On problem is, that Python is not meant to be used for real-time computing. Check [this SO answer](https://stackoverflow.com/a/24819846/15025465), where the issues are explained a little bit. PySOEM has it's use cases at (automated) testing of non-real-time dependent features of EtherCAT slaves, or some simple appliances that don't require hard real-time. Further, PySOEM can be useful for someone that tries to get familiar with the EtherCAT technology.
PySOES will have similar problems when the real-time question is asked. I'm not sure if MicroPython can be used for real-time computing, if so, looking in that direction would be worth considering.

Where to start is still a good question. It depends on where you are already good at, and which hardware you will start with...?

If it comes to writing a wrapper for a C library there are some options. When the wrapper and the library is combined into one binary (even when compling is done during the install), than you call this a binary (Python) extension. A good read is [this guide](https://packaging.python.org/guides/packaging-binary-extensions/) in the official Python packaging documentation, where the options on how the implement a binary extension are listed. For PySOEM I chose Cython. The [packaging.python.org](https://packaging.python.org) is always the first read when creating a Python package, it also covers how to upload your package to the Python Package Index (PyPI), to make you package available to `pip`.

dreaquil, 2021-02-10:
Thanks for the detailed reply @bnjmnp. My plan was to develop an EtherCAT slave on a linux machine that was running some heavy image processing. In this case, the slave response will not be real time (typically >500ms). Would it make sense to have a RaspberryPi act as the intermediary between the linux PC and the network? Considering that the application isn't real time anyway?

bnjmnp, 2021-02-11:
Okay, yes, maybe this could make sens for some use cases. The EtherCAT network could run in real time alongside your special slave that is running not in real-time. I guess your image processing slave will provide some calculated results to the master, that's fine. But beware that a traditional EtherCAT network is not meant to carry lot of image data.

bnjmnp, 2024-02-17:
I'll close this. It was good to talk about this, but would like to clean up the open issues a bit. Lets anywas try to drop a link her if anyone creates a PySOES project at some point.

closed 2024-02-17 by bnjmnp

cielavenir, 2024-03-07:
I personally searched for pysoes, but for my purpose pysoes is not what we need (I wrote the comment for my further reference).

https://github.com/OpenEtherCATsociety/SOES/tree/master/soes/hal lists up the hardware list SOES supports. It will not work on other hardwares (including normal PC NIC).

https://www.acontis.com/en/ethercat-simulation.html sells the ethercat slave simulator, but it is not free and not something I can put to CI.

For my purpose, I decided to mock master to use different communication layer.