bnjmnp/pysoem #32: Fo

ssjd2, 2021-02-12:


bnjmnp, 2021-02-13:
[image]
You can install any previous release of pysoem via `pip install pysoem==0.1.1`. FoE was introduced in 0.1.1, from there on I did only one change from 0.1.1 to 1.0.0, where I changed the parameters from `foe_write(filename, password, psize, data, timeout)` to `foe_write(filename, int password, bytes data, timeout)` (removing the psize parameter, and adding the type requirement for password and data).
Please check if the version 0.1.1 works like expected.

ssjd2, 2021-02-16:
Yes it does, v0.1.1 works like expected.

bnjmnp, 2021-02-20:
Hi @ssjd2,
I just released a new version of pysoem, where I removed the type annotation for the password. Now it may work as in version 0.1.1, though I'm still not sure why the intermediate Python `int` is a problem.
Could you please try it out, so we can close this issue!

ssjd2, 2021-02-23:
The overflow problem was fixed, thank you.

closed 2021-02-23 by ssjd2