bnjmnp/pysoem #35: can support python2.7？

touchmii, 2021-05-24:


bnjmnp, 2021-05-29:
I always hoped that I would get around this question.

I'm sure you know it: [Python 2.7 reached the end of its life on January 1st, 2020.](https://www.python.org/doc/sunset-python-2/)

But you actually could install pysoem into a Python 2.7 distribution, there are just additional requirements. Under linux you need install "python dev" see: [https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory](https://stackoverflow.com/questions/21530577/fatal-error-python-h-no-such-file-or-directory). As far as I know, under windows you need to install the Visual Studio 2010 compiler.

I never tested this.

All I can tell is, that the [examples](https://github.com/bnjmnp/pysoem/tree/master/examples) won't work because of the print's.

bnjmnp, 2022-03-02:
Assumed Solved.

I'm sure in PySOEM there won't be any efforts to support and test Python 2.7.

closed 2022-03-02 by bnjmnp