bnjmnp/pysoem #36: Slave Setup

jinkumak, 2021-06-06:
Hi,

I am new to EtherCat
I am using Stepping motor Driver **AZD2A-KED** [https://www.orientalmotor.com/products/pdfs/opmanuals/HM-60323-3.pdf](url)
My problem is that I do not understand how to implement the slave in a setup function.

Could you help me?.

Thanks

bnjmnp, 2021-06-06:
Hi @jinkumak,

sorry I cannot give very detailed support, maybe the drives vendors support team can help you. EtherCAT itself is complex, and motor control is also complex.

You basically need to know witch objects (in the index range 4000h to 4FFFh) need to be configure for the motor attached to the drive. Stuff like current limit and so on.
After that you can figure out how to actually turn the motor. Referring to the example in issue #30, you need to know how `OutputPdo` and `InputPdo` is structured, then you can control the motor with the "Controlword" and depending on the mode, "Velocity demand value" or "Position demand value".

jinkumak, 2021-06-07:
Hi @bnjmnp 

Thank you for support.

can I control the EtherCAT motor drive using a PC with Pysome program?(without EtherCAT master)

closed 2021-06-07 by jinkumak