bnjmnp/pysoem #38: UnicodeDecodeError raised by umlauts in adapter drescription.

Werdf, 2021-08-31:
https://github.com/bnjmnp/pysoem/blob/fbf7d8abc855454c0f2589d277e0830d1255ba8d/pysoem/pysoem.pyx#L80

Hello @bnjmnp,

when I call the find_adapters-function I get an UnicodeDecodeError exception, because of umlauts in the adapter drescription. 
The adapter description might be encoded in ansi.

Is it possible to remove these decode calls of name and adapter description?
or make the encoding parameterizable, because of downward compatibility?

Thank you

Greetings from Langebrück ;)

bnjmnp, 2021-09-01:
Hi @Werdf,

Thank you for this.

I guess I have expected the description to be in pure ASCII. But yes, it makes sense to return name and description as a byte array. I will figure out a solution for this.

bnjmnp, 2024-02-17:
It has bin a while, I actually solved this already in version [1.1.0](https://github.com/bnjmnp/pysoem/releases/tag/1.1.0). The description is now a raw byte array, so the decoding is up to the user.
The name will still be decoded with the hope that no one uses non-ASCII characters there.

closed 2024-02-17 by bnjmnp