bnjmnp/pysoem #43: A question of ESI and PDO

tom9672, 2021-10-18:
Hi @bnjmnp 

One thing I'm confused is how the fields mapped to the params address, it even not use any hex address. Such as 0x6040 or anything else. Actually, I'd like to know How i config the the 'inputPDO' and 'outputPDO' according to my communication object dictionary. (My device is PMM6040, with ethercat by CIA402).

My code:
```
import pysoem
import time
import ctypes
import sys

tmcm1617 = None

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('modes_of_operation_display', ctypes.c_int8),
        ('statusword', ctypes.c_uint16),
        ('position_demand_value', ctypes.c_int32),
        ('position_actual_value', ctypes.c_int32),
        ('velocity_demand_value', ctypes.c_int32),
        ('velocity_actual_value', ctypes.c_int32),
        ('torque_demand_value', ctypes.c_int32),
        ('torque_actual_value', ctypes.c_int32),
        ('digital_input', ctypes.c_uint32),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('modes_of_operation', ctypes.c_int8),
        ('controlword', ctypes.c_uint16),
        ('target_position', ctypes.c_int32),
        ('target_velocity', ctypes.c_int32),
        ('target_torque', ctypes.c_int32),
        ('digital_output', ctypes.c_uint32),
    ]

modes_of_operation = {
    'No mode': 0,
    'Profile position mode': 1,
    'Velocity mode':2,
    'Profile velocity mode': 3,
    'Homing mode': 6,
    'Cyclic synchronous position mode': 8,
    'Cyclic synchronous velocity mode': 9,
    'Cyclic synchronous torque mode': 10,
}

def convert_input_data(data):
    return InputPdo.from_buffer_copy(data)

def tmcm1617_config_func(slave_pos):
    global tmcm1617
    # All default config

def main():
    global tmcm1617
    master = pysoem.Master()
    master.open(sys.argv[1])  
    if master.config_init() > 0:
        tmcm1617 = master.slaves[0]
        tmcm1617.config_func = tmcm1617_config_func
        master.config_map()
        if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE:
            master.state = pysoem.OP_STATE
            
            master.send_processdata()
            master.receive_processdata(1_000)
            
            master.write_state()
            master.state_check(pysoem.OP_STATE, 5_000_000)
            if master.state == pysoem.OP_STATE:
                print('IN OP STATE')
                output_data = OutputPdo()
                output_data.modes_of_operation = modes_of_operation['Profile velocity mode']
                output_data.target_velocity = 500 
                for control_cmd in [6, 7, 15]:
                    output_data.controlword = control_cmd
                    tmcm1617.output = bytes(output_data)  
                    master.send_processdata()
                    master.receive_processdata(1_000)
                    time.sleep(0.01)
                try:
                    while 1:
                        master.send_processdata()
                        master.receive_processdata(1_000)
                        #  when I tried to print input values here, got error like: buffer too small, 28 not 31
                        time.sleep(0.01)
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything
                tmcm1617.output = bytes(len(tmcm1617.output))
                master.send_processdata()
                master.receive_processdata(1_000)
            else:
                print('al status code {} ({})'.format(hex(tmcm1617.al_status), pysoem.al_status_code_to_string(tmcm1617.al_status)))
                print('failed to got to op state')
        else:
            print('failed to got to safeop state')
        master.state = pysoem.PREOP_STATE
        master.write_state()
    else:
        print('no device found')
    master.close()

if __name__ == '__main__':
    main()
```
The light is blink, seems not in OP state.
I don't have a ESI file, I have a communication book, there are too many params, not sure what I need to config in 'InputPDO' and 'outputPDO'. And is there any limits of the names in the fields of InputPDO' and 'outputPDO'.

Thanks so much!
Best Regards
Tom

tom9672, 2021-10-18:
Hi @bnjmnp 
I find my ESI, could you please have a look.
And give a guidance about how to edit my inputPDO and outPDO.
Big thanks
 Limin

bnjmnp, 2021-10-21:
```python
    global pmm60
    # RatedCurrent
    pmm60.sdo_write(0x2000,0x09, bytes(0x0492))
    # MaximumCurrent
    pmm60.sdo_write(0x2000,0x0A, bytes(0x0924))
    # MaxCurrentDurTime
    pmm60.sdo_write(0x2000,0x0B, bytes(0x0064))
```
I don't understand why this stuff dose not throw an error. You send an array of 1170(0x0492), 2340, and 100 bytes to the object entries, all bytes set to zero.

Those that look like this make much more sense to me:
```python
    # Motor revolutions
    pmm60.sdo_write(0x6091,0x01, bytes(ctypes.c_uint32(1)))
```
...
Ah, now I get it, you use the empty config func. This code is not called. You better remove dead code.

bnjmnp, 2021-10-31:
if all (3) motors need to do the same move you could store the drives in a list:

```python
pmm60s = master.slaves[0:3]
```

and then:

```python
def stop_prepare_go(output_data, v, t):
    # stop and  prepare state
    output_data.Control_Word = 7
    for pmm60 in pmm60s:
        pmm60.output = bytes(output_data)  
    master.send_processdata()
    master.receive_processdata(1_000)
    time.sleep(0.01)
    # set target V
    output_data.Vi_Target_V = v
    # start state
    output_data.Control_Word = 127
    for pmm60 in pmm60s:
        pmm60.output = bytes(output_data)  
    master.send_processdata()
    master.receive_processdata(1_000)
    time.sleep(0.01)
    # go with in the setted time period
    end_t = time.time() + t
    while time.time() < end_t:
        master.send_processdata()
        master.receive_processdata(1_000)
        for i, pmm60 in enumerate(pmm60s)
            V_list[i].append(convert_input_data(pmm60.input).Velocity_actual_value)
        time.sleep(0.01)
```

Because send/receive processdata sends the data to all the drives at once, they will start nearly (but not perfectly) simultaneously.
I also removed the `sdo_read()` in the while loop an replaced it with PDO equivalent, so it's faster and you get the actual velocity from each drive at (almost) the same point in time.

tom9672, 2021-11-01:
Amazing! Thank you so much

tom9672, 2021-11-06:
Hi @bnjmnp,
Your code seems let multiple motors run together at the same velocity and time period.
My purpose is let multiple motors run at different velocity and different time period. Some times their working period may overlapping. 
The previous function “stop_prepare_go()”  unable to achieve overlapping, it can only let the motors run at the same period.
Actually my machine have three axis, with three servo motors. They work at different status. Sometimes, the start of one motor depends on another.
Do you have any suggestions for this.
Hope you have a good day.
Looking forward to hearing from you.
Tom

tom9672, 2021-11-15:
Hi @bnjmnp 
Could you please have a look my previous question when you are free.
Thanks so much!!!
Best regards,
Tom

bnjmnp, 2021-11-16:
Hi @tom9672,
Sorry for the late response. I also thought I give you some time to figure this out yourself.
If you want the motors run differently, you may use [threading](https://docs.python.org/3/library/threading.html).
First thread you would use for `send_processdata()` and `receive_processdata(1_000)` like in the [basic_example.py](https://github.com/bnjmnp/pysoem/blob/074b8d7e12e27dac9d23cbbca62d91d39625327a/examples/basic_example.py) (see the `_processdata_thread()`).
Then it is fully up to you, you may spin up a thread from the main context for every job a motor needs to do, or you start a thread for each motor at the beginning. You could also write you own state machine and do the switching between the motors yourself.

closed 2022-04-07 by tom9672

tdemierr, 2023-04-12:
Hello,
Sorry to up this thread again. However, I am currently interfacing this drive: http://www.rtelligent.net/upload/wenjian/Stepper/ECT%20Series%20User%20Manual.pdf 
Could it be possible to share the version of your code with threading ?
Thanks