bnjmnp/pysoem #44: ifm AL1332 EtherCAT slave - Cannot get OP state

mandeepahuja1, 2021-12-10:
Hello team,
I’m using the pysoem library and trying to communicate with the ifm AL1332 EtherCAT slave. I have spent ample amount of time learning and reading all the examples and issues under this GitHub repository. 

I have successfully been able to do an SDO read of various indexes on the slave. 

I am unable to get the device to SAFEOP or OP State, hence I am unable to do any outputs or cyclic process data exchange. 

Do I HAVE to run config_func and config_map when all I am doing is using default setup? If yes, what goes in them?

Also, I’m just going to throw this out there I’m willing to pay for someone’s time to get this working with me. All I want to do is get a working cyclic exchange of data with the AL1332. 

Please advice.

mandeepahuja1, 2021-12-10:
Tagging @bnjmnp

bnjmnp, 2021-12-11:
Hi @mandeepahuja1,

If the device is not even able to reach SafeOP state, it is very likely that there are configurational issues.

What does the "al status code" tell?
[https://github.com/bnjmnp/pysoem/blob/27091cdb415f6ab4b1e3a5c6c36bd0c403ab360f/examples/minimal_example.py#L58-L66](https://github.com/bnjmnp/pysoem/blob/27091cdb415f6ab4b1e3a5c6c36bd0c403ab360f/examples/minimal_example.py#L58-L66)

I took a quick look at the manual of the device, maybe it is necessary to configure the device via the IoT interface first.

You definitely need to call `config_map()`. With it the devices are pushed to SafeOP state in a proper way. You may or may not provide a "config_func" for your device (like the `el3002_setup()` function in the minimal_example.py), this is depends on the device, some require configurational steps others not. A provided "config_func" is called during the execution of `config_map()`.

mandeepahuja, 2021-12-12:
Hello @bnjmnp - I ran the code snippet you mentioned in the previous post and here is the result

> ifm IO-Link Master did not reach SAFEOP state
> al status code 0x1e (Invalid input configuration)

Also, I noticed when I run master.config_map(), the ERR Led on the AL1332 starts blinking. If I don't run the config_map() then I can do SDO_Read's fine. 

According to the manual 
ERR Led blinking means "Invalid Configuration"
RUN Led blinking means "Device in PRE-OPERATIONAL STATE" 

How do I setup this input configuration? I mean what would that entail. I don't have a config_func(). I don't have a config_func() because I don't know what goes in there :). 

Here is a screenshot from the CODESYS IDE where everything works as expected. 
[image]

No settings are required to be changed from the IoT port. It's just default configuration. 

I read one of your posts before that getting to OP_STATE is first step. Well I need your help getting to SAFE_OP and then OP_STATE. 

what should I do next? Yes, I understand I need a bit of hand holding. 

Also, When I read the following
```
rxPDOAssign = device.sdo_read(0x1C12, 1)
txPDOAssign = device.sdo_read(0x1C13, 1)
```
The output looks correct as shown in the manual
> 0x1600
> 0x1A81

bnjmnp, 2021-12-12:
Could you just try to run CODESYS IDE and got to OP state, and then go to PreOP. And now connecting with your pysoem system, without a power cycle or reset to the AL1332. And then see if your script without "config_func" works. Just to see that the configurational steps done in the CODESYS IDE is the missing piece in the pysoem script.

bnjmnp, 2021-12-14:
Sorry, I'm verry busy thees days so sending me one of the AL1332 is not the best option.
Could you take a screenshot of the "Startup Parameters" probably there are the things you need to put in the "config_func".
[image]

mandeepahuja, 2021-12-15:
Hello @bnjmnp 
You might be spot on here
[image]

All these parameters make sense to me. I mean I understand what they are. 

I also captured a Wireshark trace of every CoE request done by CODESYS. 

Can you help me with one example of config_func() using the above screenshot from Startup Parameters? Please and Thank you!

bnjmnp, 2021-12-15:
This should do it:
```py
def config_func_imf_al1332(device_pos)
    device = ...
    for i in range(8):
        device.sdo_write(0x8000+i*0x10, 0x24, bytes([159]))
        device.sdo_write(0x8000+i*0x10, 0x25, bytes([159]))
        device.sdo_write(0x8000+i*0x10, 0x28, (3).to_bytes(2, byteorder='little', signed=False)) # 0: Deactivated; 1: Digital Input; 2: Digital Output; 3: IO-Link

    device.sdo_write(0xF030, 0x00, bytes([0]))
    for i in range(8):
        device.sdo_write(0xF030, 0x01+i, (4882).to_bytes(4, byteorder='little', signed=False))
    device.sdo_write(0xF030, 0x00, bytes([8]))
```
I used a loop as all 8 ports are configured the same.

mandeepahuja, 2021-12-15:
Thank You @bnjmnp 
At a minimum I had to add the mappings to output 0x1c12. The inputs I did not have to setup. 
`device.sdo_write(0x1C12, 0x00, bytes([0]))
    for i in range(8):
        device.sdo_write(0x1C12, 0x01+i, (0x1600+i).to_bytes(2, byteorder='little', signed=False))

    device.sdo_write(0x1C12, 0x00, bytes([8]))`

Lessons learned:

1. The esi file contains information on how to setup the EtherCAT slave. However deciphering the esi manually is not possible so using Documentation and CODESYS helped understand what all parameters are needed to be set to get everything working.
2. If you cannot get to SAFE_OP or OP_STATE that simply means your config_func() is missing things that are needed to setup the slave correctly. Also you may need to look at how fast you poll the process data. 

This issue can be closed.

Thank you again @bnjmnp for your excellent support.

bnjmnp, 2021-12-16:
You are welcome.

closed 2021-12-16 by bnjmnp