bnjmnp/pysoem #47: AttributeError: 'pysoem.pysoem.CdefSlave' object has no attribute 'pos'

tom9672, 2022-02-15:
https://pysoem.readthedocs.io/en/latest/cdef_slave.html#

as the document, slave should have the attr of pos, my pysoem version is 1.0.3.

bnjmnp, 2022-02-15:
You are right, there is a mistake in the docs.
I'll look into it soon.

Thank you for your report.

bnjmnp, 2022-02-20:
Can you think of a situation where the `pos` attribute could be helpful? I tend to just remove it from the docs.
If you loop through the slaves list you could get the position of the slave by the list index.
```python
    for i, slave in enumerate(master.slaves):
        print(f'Found device {slave.name} in position {i}.')
```

tom9672, 2022-02-23:
The ```pos``` attribute is not important