bnjmnp/pysoem #48: Installation Fails (unable to open wpcap.lib) if using Python 3.10

jkwphysics, 2022-02-21:
I upgraded my python installation a few days ago to a new directory, and ran into this issue when installing pysoem.

```
C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:.\soem\oshw\win32\wpcap\Lib\x64 /LIBPATH:C:\Users\*******\AppData\Local\Programs\Python\Python310\libs /LIBPATH:C:\Users\*******\AppData\Local\Programs\Python\Python310\PCbuild\amd64 /LIBPATH:C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.31.31103\lib\x64 /LIBPATH:C:\Program Files (x86)\Windows Kits\10\lib\10.0.19041.0\ucrt\x64 /LIBPATH:C:\Program Files (x86)\Windows Kits\10\\lib\10.0.19041.0\\um\x64 wpcap.lib Packet.lib Ws2_32.lib Winmm.lib /EXPORT:PyInit_pysoem build\temp.win-amd64-3.10\Release\.\soem\osal\win32\osal.obj build\temp.win-amd64-3.10\Release\.\soem\oshw\win32\nicdrv.obj build\temp.win-amd64-3.10\Release\.\soem\oshw\win32\oshw.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatbase.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatcoe.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatconfig.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatdc.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatfoe.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatmain.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatprint.obj build\temp.win-amd64-3.10\Release\.\soem\soem\ethercatsoe.obj build\temp.win-amd64-3.10\Release\pysoem/pysoem.obj /OUT:build\lib.win-amd64-3.10\pysoem\pysoem.cp310-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.10\Release\.\soem\osal\win32\pysoem.cp310-win_amd64.lib
      LINK : fatal error LNK1181: cannot open input file 'wpcap.lib'
      error: command 'C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.31.31103\\bin\\HostX86\\x64\\link.exe' failed with exit code 1181
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pysoem
```

I do have npcap installed in wpcap compatibility mode.  I uninstalled my existing pysoem module in my python 3.9 directory and reinstalled it with no errors, so it seems restricted to the 3.10 version of python.

bnjmnp, 2022-02-22:
Hi. At the moment I provide binary wheels only up to Python 3.9 for Windows. This makes installation a lot less painful. For Python 3.10 when pip installing pysoem, pip tries to compile a wheel from the sources. Even though you have the VS compiler installed the compile dose not get through. I will take a closer look at this on the coming weekend. I already was about to release a new version of pysoem because of a small new feature, so at the same time I can also take care of Python 3.10 and provide a 3.10 wheel for the new release.

bnjmnp, 2022-02-27:
I've just created a new release that includes a binary wheel for Python 3.10 for windows. Are you able to install pysoem now?

jkwphysics, 2022-02-27:
> I've just created a new release that includes a binary wheel for Python 3.10 for windows. Are you able to install pysoem now?

I just installed 1.0.4 with Python 3.10.2 with no issues. Did not need to compile anything.  Thanks very much!

closed 2022-02-27 by jkwphysics