bnjmnp/pysoem #50: How to write PDO mapping config

tom9672, 2022-02-27:
We can use the default pdo mapping by this:
```
class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ =[
        ('Control_Word',ctypes.c_uint16),
        ('Mode_Operation',ctypes.c_int8),
        ('Target_Position',ctypes.c_int32),
        ('Touch_probe_function',ctypes.c_uint16),
        ('physical_outputs',ctypes.c_uint32),
        ('bit_mask',ctypes.c_uint32)
    ]   

output_data = OutputPdo()
```
but how can we set the pdo mapping (write pdo mapping to device)?

I guess, use sdo to write to the address x1A00, x1600?  but the pdos are lists of addresses, how to set the PDOS？ so that i can select the pdos that i need instead of using the default.

rako233, 2022-02-27:
Take a look into CanOpen specs, how to to this. It doesn't work with all slaves though, because they can't be changed

bnjmnp, 2022-02-27:
I think in CANOpen it is not the best direction to look into, there it is a bit different.

For each EtherCAT PDO change might be implement in another way.

* (fixed PDO mapping:) For some device changing PDO mapping is not possible at all, this is the case when 0x1C12/0x1C13 and 0x1600-0x17FF/0x1A00-0x1BFF are read-only .
* (Selectable PDO mapping:) Others allow PDO change by allowing write access to 0x1C12/0x1C13 where only 0x1600-0x17FF/0x1A00-0x1BFF is read only. This is maybe the most common implementation.
Examples are shown in the `elxxxx_setup` functions of the [minimal_example.py](https://github.com/bnjmnp/pysoem/blob/31489b7d673eccaf72fe40761bbb47d66d854119/examples/minimal_example.py#L31) and [basic_example](https://github.com/bnjmnp/pysoem/blob/31489b7d673eccaf72fe40761bbb47d66d854119/examples/basic_example.py#L39)
* (Configurable PDO mapping:) Some device allwo complete PDO change by allowing write access to both 0x1C12/0x1C13 and 0x1600-0x17FF/0x1A00-0x1BFF.
* (?:) Devices that are similar to CANopen allow write access to only 0x1600/0x1A00, where 0x1C12/0x1C13 are read only or 0x1C12/0x1C13 are only allowed to map 0x1600/0x1A00 or nothing at all.

closed 2022-04-07 by tom9672