bnjmnp/pysoem #52: On Windows, after master.config_init() a short sleep() has to be done

rako233, 2022-02-27:
I tried to use the pysoem on Python 3.10 and 3.9 for Windows but the issue was the same, when not having a sleep(0.1) after master.config_init() the results are garbage. On Linux this is not an issue.

bnjmnp, 2022-02-27:
Hi. Do you like to share more context on this? For example what your script is doing right after the `config_init()`, what the error looks like, or maybe even more...
Here and there sleeps can help, maybe sometimes an increased timeout will do the same.

rako233, 2022-03-04:
I  did this and a Timeout didn't helped

`    def open(self, interface) : 
        self.master.open(self.interface)
        if self.master.config_init() > 0 :
            sleep(0.1)
        else:
            raise EthercatException()
`