bnjmnp/pysoem #55: How to log to console?

AKAMEDIASYSTEM, 2022-03-05:
I'm sorry to have such a basic question, but I'm trying to step through the pyx file to determine where a particular sdo write is occurring (see #53), and I'm unable to get log messages to appear while I'm doing this.
If I'm running `basic_example.py` and I want to see all the log messages emitted by the .pyx file, what should I do? I've tried various ways that I usually involve adding something like `logging.basicConfig(stream=sys.stdout, level=logging.INFO)` to the head of my script (I've also added the same line to the head of the pyx file), but neither seem to have a result.
I don't usually get this far into the weeds when I use a module, so again my apologies if I'm asking a basic question here, I've searched for "how to see log messages from cython" and "how to see log messages from imported module" to no avail...
Thanks for any help/hint you can offer!

bnjmnp, 2022-03-06:
No worries, this is all fine. The way you set up logging looks good.
On SDO read and write nothing is logged inside pysoem. Within `config_map()` only the IO map size returned by SOEMs `ecx_config_map_group()` is logged. So when `config_map()` is called you would only see the `io map size: x` log message.
Even with logging added to pysoems SDO read and write, the SDO read and writes that the underlying SOEM is doing inside `ecx_config_map_group()` won't get logged.
For debugging the best would be to have a Wireshark record.

bnjmnp, 2022-03-06:
Wait. You need to set logging level to logging.DEBUG to see the `io map size: x` message.

AKAMEDIASYSTEM, 2022-03-06:
Thanks! I think this issue can be closed, and yeah, it's looking like I need to be digging into SOEM itself to figure out what's going on...

closed 2022-03-06 by AKAMEDIASYSTEM