bnjmnp/pysoem #59: Can't Install on M1 mac

michaelschem, 2022-04-12:
I get the following error when I try to install on mac.

```
(venv) mschem@mbp sk % pip install pysoem==1.0.4
Collecting pysoem==1.0.4
  Using cached pysoem-1.0.4.tar.gz (342 kB)
    ERROR: Command errored out with exit status 1:
     command: /Users/mschem/sk/venv/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/48/_sj4kz4d2qn3xdqrkln2gfmc0000gn/T/pip-install-ycddbiht/pysoem_f4a291dcd77f46ccaea12e44a1752d73/setup.py'"'"'; __file__='"'"'/private/var/folders/48/_sj4kz4d2qn3xdqrkln2gfmc0000gn/T/pip-install-ycddbiht/pysoem_f4a291dcd77f46ccaea12e44a1752d73/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/48/_sj4kz4d2qn3xdqrkln2gfmc0000gn/T/pip-pip-egg-info-wbe5oi44
         cwd: /private/var/folders/48/_sj4kz4d2qn3xdqrkln2gfmc0000gn/T/pip-install-ycddbiht/pysoem_f4a291dcd77f46ccaea12e44a1752d73/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/48/_sj4kz4d2qn3xdqrkln2gfmc0000gn/T/pip-install-ycddbiht/pysoem_f4a291dcd77f46ccaea12e44a1752d73/setup.py", line 32, in <module>
        soem_macros.append(('EC_VER2', ''))
    NameError: name 'soem_macros' is not defined
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/b4/92/52b9c1b239b7ccc8c7032ddab1658029e3326f7b59099c2474f4d090b02c/pysoem-1.0.4.tar.gz#sha256=ae4993de7508f24467543f59a801819b64c84dcbe11df3b1de70e39415693125 (from https://pypi.org/simple/pysoem/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pysoem==1.0.4 (from versions: 0.0.13, 0.0.14, 0.0.15, 0.0.16, 0.0.17, 0.0.18, 0.1.1, 1.0.0, 1.0.1, 1.0.2, 1.0.3, 1.0.4)
ERROR: No matching distribution found for pysoem==1.0.4
WARNING: You are using pip version 21.1.2; however, version 22.0.4 is available.
You should consider upgrading via the '/Users/mschem/sk/venv/bin/python -m pip install --upgrade pip' command.
```

I noticed that `sys.platform` returns `darwin` for mac, so I've added the following to the setup.py. I'm using macosx as I see that folder in `soem/osal` and it seems to be the right place.

```
elif sys.platform.startswith('darwin'):
    soem_macros = []
    soem_lib_dirs = []
    soem_libs = ['pthread', 'rt'] 
    os_name = 'macosx'
```

Now I get down to:

```
10.14-arm64-3.8/./soem/soem/ethercatmain.o build/temp.macosx-10.14-arm64-3.8/./soem/soem/ethercatprint.o build/temp.macosx-10.14-arm64-3.8/./soem/soem/ethercatsoe.o -lpthread -lrt -o build/lib.macosx-10.14-arm64-3.8/pysoem/pysoem.cpython-38-darwin.so
ld: library not found for -lrt
clang: error: linker command failed with exit code 1 (use -v to see invocation)
error: command 'clang' failed with exit status 1
```

I'm working on figuring out what's going on here, but I figured I'd open an issue to see if smarter minds than my own know what's going on here.

Also worth noting that I can build the c SOEM package:

```
build % cmake ..
-- The C compiler identification is AppleClang 13.0.0.13000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- OS is macosx
-- LIB_DIR: lib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mschem/SOEM/build
mschem@mbp build % make
[  4%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.o
[  9%] Building C object CMakeFiles/soem.dir/soem/ethercatcoe.c.o
[ 14%] Building C object CMakeFiles/soem.dir/soem/ethercatconfig.c.o
[ 19%] Building C object CMakeFiles/soem.dir/soem/ethercatdc.c.o
[ 23%] Building C object CMakeFiles/soem.dir/soem/ethercateoe.c.o
[ 28%] Building C object CMakeFiles/soem.dir/soem/ethercatfoe.c.o
[ 33%] Building C object CMakeFiles/soem.dir/soem/ethercatmain.c.o
[ 38%] Building C object CMakeFiles/soem.dir/soem/ethercatprint.c.o
[ 42%] Building C object CMakeFiles/soem.dir/soem/ethercatsoe.c.o
[ 47%] Building C object CMakeFiles/soem.dir/osal/macosx/osal.c.o
[ 52%] Building C object CMakeFiles/soem.dir/oshw/macosx/nicdrv.c.o
[ 57%] Building C object CMakeFiles/soem.dir/oshw/macosx/oshw.c.o
[ 61%] Linking C static library libsoem.a
[ 61%] Built target soem
[ 66%] Building C object test/simple_ng/CMakeFiles/simple_ng.dir/simple_ng.c.o
[ 71%] Linking C executable simple_ng
[ 71%] Built target simple_ng
[ 76%] Building C object test/linux/slaveinfo/CMakeFiles/slaveinfo.dir/slaveinfo.c.o
[ 80%] Linking C executable slaveinfo
[ 80%] Built target slaveinfo
[ 85%] Building C object test/linux/eepromtool/CMakeFiles/eepromtool.dir/eepromtool.c.o
[ 90%] Linking C executable eepromtool
[ 90%] Built target eepromtool
[ 95%] Building C object test/linux/simple_test/CMakeFiles/simple_test.dir/simple_test.c.o
[100%] Linking C executable simple_test
[100%] Built target simple_test
```

tom9672, 2022-04-13:
works on Linux and windows, I also tried on a Mac, but it seems not work. Have you tried on a virtual machine on mac?

michaelschem, 2022-04-13:
I have not tried on a virtual machine for mac yet. I was thinking the problem was probably with my chip architecture since I'm on an M1.  I'll try in a virtual machine and update here.

tom9672, 2022-04-13:
I tried on a Mac x86 interl i5 cpu, also not work. But definitely works well on Linux and Windows.

raveslave, 2023-07-04:
pip install pysoem on Intel Mac also results in    
`No matching distribution found for pysoem`

any tricks to roll this on mac? (no VM)

Ice-Panda, 2023-11-09:
@tom9672 
Hi, I tried on a Mac x86 intel i7 cpu, and it works well for me.
What I did was download the source code and add some code for platform adaptation.
```python
elif sys.platform.startswith('darwin'):
    soem_macros = []
    soem_lib_dirs = []
    soem_libs = ['pthread', 'pcap']
    os_name = 'macosx'
```
Then, you can use the following command to install the pysoem library in the same directory as the `setup.py` file. 
```bash
pip install .
```
[image]
ps: I changed the version number to 1.1.5.
[image]

Oh, If you find that `master.open(ifname)` fails, please modify the read and write permissions of the mac's network port. 
For example, I use the following command to open the permissions of four thunderbolt ports.
```bash
sudo chmod 777 /dev/bpf*
```
Enjoy it!

bnjmnp, 2023-11-11:
Thank you @Ice-Panda for sharing, this was already available on a branch: [https://github.com/bnjmnp/pysoem/blob/add_macos/setup.py](https://github.com/bnjmnp/pysoem/blob/add_macos/setup.py)
I just merged it into mainline. Now also binary wheels are build for macOS, so no build step is needed during the install.
With that I think we can close this now.

closed 2023-11-11 by bnjmnp