bnjmnp/pysoem #62: Retrieval of Assigned Name SDO returns different bytes every try

njoye, 2022-04-29:
Hi!

I've used the following code to retrieve the "Assigned Name" of a Synapticon Motor Controller. Sadly I always get a different result when I retrieve the data (only once was I able to actually decode the name and get the correct answer). When executed, I usually get a `UnicodeDecodeError`. The decoding doesn't seem to be the actual issue as the encoding provided by the company (ASCII) did work once and works with lower-level approaches (namely Igh and TwinCAT) of EtherCAT masters. However I am wondering why I get different results every time I try to load the data - is this a bug in the software? Maybe with how I set up my hardware (I communicate with other devices on the same interface - is this an issue)? I thank you for any help you can give me already!

The code that shows my problem:
```
import pysoem

def find_devices(ifname):
        master = pysoem.Master()
        master.open(ifname)

        if master.config_init() > 0:
                for device in master.slaves:
                        print("Found device " + str(device.name))
                         # The below call ('decode') makes it crash as sdo_read returns different values every time
                        device_name = device.sdo_read(0x20F2, 0).decode('ascii') 
                        print("Device name is " + str(device_name))
        else:
                print("No device found")

        devices = master.slaves
        master.close()
        return devices

if __name__ == "__main__":
        find_devices("enp8s0")
```

bnjmnp, 2022-04-30:
Hi,
What do you mean with "I communicate with other devices on the same interface"? On the Ethernet port you are using, there should only be EtherCAT device connected.

njoye, 2022-05-02:
Thanks @bnjmnp - after doing my own research I came to the same conclusion myself. I'll talk to our electrical engineer and make sure that the electrical wiring is correct. Once that is done and if that fix resolves this issue I will close it. Thanks again!

njoye, 2022-05-11:
The wiring is correct now, but my issues with reading the data persists. This was also confirmed by a developer at Synapticon who had the same problem with the script that I provided above.