bnjmnp/pysoem #64: I am using EK1100, EL2008 module. and I want to light the EL2008 with an LED.

jehun0908, 2022-05-09:
I am using EK1100, EL2008 module. I want to light the EL2008 with an LED.
In this case, do I need a safe_op or an open state?
Also, if necessary, is it operated in the order of PreOP, SafeOP, and OPstate?

>
 import sys
import time
import struct
import pysoem
import chardet
import ctypes
import struct

master = pysoem.Master()

master.open('NPF_{F0A11C28-9838-4230-8341-3FA8D32CD718}')
print("EtherCAT master created and started...")

if master.config_init() > 0:
    print("slaveopen")
    
    master.read_state()
    print("slave status")
    print(master.read_state())
    
    for slave in master.slaves:
        
        print('{}:'.format(slave.name))                       
        print(chardet.detect(slave.name.encode()))            
        print("--------slavename--------")
    
    slave0 = master.slaves[0]
    slave1 = master.slaves[1]
    slave0.state
    slave1.state
    slave0.write_state
    slave1.write_state
    slave1.sdo_write(0x7050, 1, bytes(ctypes.c_bool(1)))
    if slave1.state == pysoem.OP_STATE :
        print("True")    
    else :
        print("False")
    master.send_processdata()                                           # send Data
    master.receive_processdata(1_000)                                   # recieve Data
    slave1.output = bytes(len(slave1.output))
else :
    print("No device found")   
master.close()

print("------------------------------------End--------------------------------")

bnjmnp, 2022-05-16:
I think OP state is necessary to change the states of outputs. You probably also need to toggle the output via PDO communication like in the [basic_example.py](https://github.com/bnjmnp/pysoem/blob/0b6d2d30bb35b5acc562e86fa8ee96aecb0de0c5/examples/basic_example.py#L74).
PreOP, SafeOP and then OP state is the right sequence, yes.

jehun0908, 2022-05-17:
I turned on the LED referring to basic_example.py.
This is a reference code for other people.
'''

import sys
import struct
import time
import threading

from collections import namedtuple

import pysoem

class BasicExample:

    BECKHOFF_VENDOR_ID = 0x0002
    EK1100_PRODUCT_CODE = 0x044c2c52
    EL2008_PRODUCT_CODE = 0x7d83052

    def __init__(self, ifname):
        self._ifname = ifname
        self._pd_thread_stop_event = threading.Event()
        self._ch_thread_stop_event = threading.Event()
        self._actual_wkc = 0
        self._master = pysoem.Master()
        self._master.in_op = False
        self._master.do_check_state = False
        SlaveSet = namedtuple('SlaveSet', 'name product_code config_func')
        self._expected_slave_layout = {0: SlaveSet('EK1100', self.EK1100_PRODUCT_CODE, None),
                                       1: SlaveSet('EL2008', self.EL2008_PRODUCT_CODE, None)}

    
    def _processdata_thread(self):
        while not self._pd_thread_stop_event.is_set():
            self._master.send_processdata()
            self._actual_wkc = self._master.receive_processdata(10000)
            if not self._actual_wkc == self._master.expected_wkc:
                print('incorrect wkc')
            # time.sleep(0.01)

    def _pdo_update_loop(self):

        self._master.in_op = True

        output_len = len(self._master.slaves[1].output)

        tmp = bytearray([0 for i in range(output_len)])

        toggle = True
        try:
            while 1:
                if toggle:
                    tmp[0] = 0x00
                else:
                    tmp[0] = 0xFF
                self._master.slaves[1].output = bytes(tmp)

                toggle ^= True

                # time.sleep(1)

        except KeyboardInterrupt:
            # ctrl-C abort handling
            print('stopped')

    def run(self):

        self._master.open(self._ifname)

        if not self._master.config_init() > 0:
[... 90 lines cut ...]
                for i, slave in enumerate(self._master.slaves):
                    if slave.state != pysoem.OP_STATE:
                        self._master.do_check_state = True
                        BasicExample._check_slave(slave, i)
                if not self._master.do_check_state:
                    print('OK : all slaves resumed OPERATIONAL.')
            time.sleep(0.01)

class BasicExampleError(Exception):
    def __init__(self, message):
        super(BasicExampleError, self).__init__(message)
        self.message = message

if __name__ == '__main__':
    
    print('basic_example started')

    ifname = '[Your Device Name]'

    try:
        BasicExample(ifname).run()
    except BasicExampleError as expt:
        print('basic_example failed: ' + expt.message)
        sys.exit(1)
'''

jehun0908, 2022-05-17:
I'd like to make a motor move separately from the above code.
I faced one difficulty.
'''
if master.state_check(pysoem.SAFEOP_STATE, 50_000) == pysoem.SAFEOP_STATE: 
'''
If you print from the code inside "",

Master.state_check(pysoem.SAFEOP_STATE) outputs 2 values, and pysoem.SAFEOP_STATE outputs 4 values.
So, we are not able to enter the SAFEOP state. Can you help me?

bnjmnp, 2022-05-17:
BTW thiw is the way to generate code blocks on GitHub:
~~~
```py
def foo():
    pass
```
~~~
This will result in syntax highlighted code blocks:
```py
def foo():
    pass
```

bnjmnp, 2022-05-17:
In #30 there is an example that brings a motor of a servo drive to move.
If you cannot reach SafeOP state, there is probably an configuration issue, it is good to read out the AL status code like in the [minimal_example.py](https://github.com/bnjmnp/pysoem/blob/0b6d2d30bb35b5acc562e86fa8ee96aecb0de0c5/examples/minimal_example.py#L64)

jehun0908, 2022-05-17:
I looked at #30 and referred to it.
Al status code indicates 0x0 (No error)
I cannot reach SafeOP state.....

```py
import pysoem
import time
import ctypes
import struct

KINBT11 = None

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('Statusword', ctypes.c_uint16),
        ('Actual_Postion', ctypes.c_int32),
        ('Actual_Velocity', ctypes.c_int32),
        ('Digital_Input', ctypes.c_uint16),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('Target_Velocity', ctypes.c_int32),
        ('Controlword', ctypes.c_uint16),
    ]

def config_func(slave_pos):
    global KINBT11
    
    # encoder increments
    KINBT11.sdo_write(0x608F, 1, bytes(ctypes.c_uint32(500)))
    # Motor revolutions
    KINBT11.sdo_write(0x608F, 2, bytes(ctypes.c_uint32(1)))
    
    # Max profile velocity
    KINBT11.sdo_write(0x607F, 0, bytes(ctypes.c_uint32(8000)))
    # Max motor speed
    KINBT11.sdo_write(0x6080, 0, bytes(ctypes.c_uint32(8000)))
    
    # # gear ratio
    KINBT11.sdo_write(0x6091, 1, bytes(ctypes.c_uint32(1)))
    KINBT11.sdo_write(0x6091, 2, bytes(ctypes.c_uint32(91)))    

def convert_input_data(data):
    return InputPdo.from_buffer_copy(data)

def main():
    global KINBT11
    master = pysoem.Master()
    master.open('[My device name]')                       
    if master.config_init() > 0:                                                    
        KINBT11 = master.slaves[0]                                                
        KINBT11.config_func = config_func                                 
        master.config_map()                                                        
        print(master.state_check(pysoem.SAFEOP_STATE,50000))                         #### The value appears as 2.
        print(pysoem.SAFEOP_STATE)                                                   #### The value appears as 4.
        if master.state_check(pysoem.SAFEOP_STATE, 50000) == pysoem.SAFEOP_STATE:    # From here, we keep going to the else 
            print("True")
            master.state = pysoem.OP_STATE                                         
            master.write_state()                                                   
            master.state_check(pysoem.OP_STATE, 5000000)                          
            if master.state == pysoem.OP_STATE:                                    
                output_data = OutputPdo()                                           
                output_data.Target_Velocity = 500                                      
                output_data.Controlword = 15                                 
                KINBT11.output = bytes(output_data)                   
                master.send_processdata()                                     
                master.receive_processdata(1000)                              
                time.sleep(0.01)
                try:
                    while 1:
                        master.send_processdata()                                
                        master.receive_processdata(1000)                         
                        print(convert_input_data(KINBT11.input).Target_Velocity)
                        print(convert_input_data(KINBT11.input).Statusword)   
                        time.sleep(0.01)
                except KeyboardInterrupt:
                    print('stopped')
                # zero everything 
                KINBT11.output = bytes(len(KINBT11.output))
                master.send_processdata()                                         
                master.receive_processdata(1000)                                
            else:
                print('failed to got to op state')
                print('al status code {} ({})'.format(hex(KINBT11.al_status), pysoem.al_status_code_to_string(KINBT11.al_status)))
        else:
            print('failed to got to safeop state')
            print('al status code {} ({})'.format(hex(KINBT11.al_status), pysoem.al_status_code_to_string(KINBT11.al_status)))
        master.state = pysoem.PREOP_STATE
        master.write_state()
    else:
        print('no device found')
    master.close()

if __name__ == '__main__':
    main()
``
![RxPDO](https://user-images.githubusercontent.com/84609465/168768874-b9209a16-45c7-4d9e-a0b7-8c608bab32da.PNG)
![TxPDO](https://user-images.githubusercontent.com/84609465/168768889-ef015e62-c5e0-45bb-8117-cc84efae022d.PNG)
`

bnjmnp, 2022-05-18:
You probably tried to increase the timeout for the `state_check`?
And you are sure the KINBT11 is the only device in the EtherCAT network?

jehun0908, 2022-05-18:
1.  I tried increasing "Timeout of State_check" but there was no change.
2. The KINBT11 driver is a driver created using the ET1100 and works well using the TwinCAT program.
But I want to use Python to operate it.
3. This is the only driver I have.

'''py
if master.state_check(pysoem.SAFEOP_STATE, 50000) == pysoem.SAFEOP_STATE:
'''
When outputting the value "if master.state_check(pysoem.SAFEOP_STATE, 50000" is 2 and go to the else statement.

jehun0908, 2022-05-18:
Accept 0x1A02 value from inputPdo instead of default 0x1A00. Where should I add this part?
[image]

bnjmnp, 2022-05-18:
Can you pleased also take a screenshot of the Startup tab? There you will see what sdo_writes are needed in the config function.

jehun0908, 2022-05-18:
I'm not sure if the picture of start_tab you want is correct.

[image]

bnjmnp, 2022-05-18:
The one next to this one, on the left.

jehun0908, 2022-05-19:
Oh! Startup_Tab is here.

[image]

bnjmnp, 2022-05-19:
Right, this is what TwinCAT is doing at the transition from PreOP to SafeOP state.

The equivalent place to put that is in the "config function", because the code there is also called in the transition from PreOP to SafeOP (during the `config_map()`)

You could do that ether this way:

```py
def config_func(slave_pos):
    global KINBT11
    rx_map_obj_bytes = struct.pack('BxH', 1, 0x1602)
    KINBT11.sdo_write(0x1c12, 0, rx_map_obj_bytes, True)
    tx_map_obj_bytes = struct.pack('BxH', 1, 0x1A02)
    KINBT11.sdo_write(0x1c13, 0, tx_map_obj_bytes, True)
```

or that way:

```py
def config_func(slave_pos):
    global KINBT11
    KINBT11.sdo_write(0x1c12, 0, bytes([0x01, 0x00, 0x02, 0x16]), True)
    KINBT11.sdo_write(0x1c13, 0, bytes([0x01, 0x00, 0x02, 0x1A]), True)
```

Not sure if this is 100% correct.
I'm also not sure it this will solve the problem, but it is worth to try it.

jehun0908, 2022-05-19:
Thank you for your time on my issue.
I changed it according to your advice.
But there is still a problem.
The result value of the code below is still 2.
I am interpreting "master.state_check(pysoem.SAFEOP_STATE, 50000)" as waiting for the master to reach the state of "SAFEOP_STATE"

Is there a way to do it without the code below?

```py
master.state_check(pysoem.SAFEOP_STATE, 50000) == pysoem.SAFEOP_STATE
```

jehun0908, 2022-05-19:
Accessing COE objects with sdo_write and sdo_read works fine.
But I still can't change to OP_STATE.

jehun0908, 2022-05-20:
oh.. I thought the computer read the COE without any problems.
However, if you try to read coe after config_map(), an error is printed.
Is this because of the error of misreading eeprom?

bnjmnp, 2022-05-20:
Could you post the error message?
Maybe the drive also needs DC (Distributed Clocks) configured. You can check the DC tab in TwinCAT or take a look at the ESI file.

jehun0908, 2022-05-20:
This is the error message and DC tab you mentioned.
max_filename_speed1 is output. However, an error occurs when outputting again after config_map().

[image]

[image]

bnjmnp, 2022-05-20:
Okay DC is not needed, this is fine.

For some reaseon the drive issues an Emergency Error. The 40960 is hex 0xA000. You need to check the manual what might cause this Emergency Error or ask the vendor of you device.

jehun0908, 2022-05-20:
Thank you for your advice. I will look at the manual and ask the supplier according to your advice.
There are still two questions left.

1. Is it right that "config_map()" does not affect sdo_read?
2. Is there a high probability that "master.state_check(pysoem.SAFEOP_STATE,50000)" represents a value of 2 because of the device?

bnjmnp, 2023-01-03:
It has been a while. Sorry for not ansering to your last post, somehow I lost track of this conversation. Do you still have issues with this? Did you had any success in the end?

bnjmnp, 2024-02-03:
Closed due to inactivity of the OP. Please reopen the ticket if you need further help on this.

closed 2024-02-03 by bnjmnp