bnjmnp/pysoem #65: Toggling a LED with EL2809 and pysoem

MarcelloDavis, 2022-05-10:
Hello,

I am new to EtherCAT and pysoem. I installed the library on a raspberry Pi and am able to detect the connected slave devices (EK1100 and EL2809). An LED is connected to the EL2809 on output 1. So my goal is to toggle the LED via the EL2809 using pysoem on the raspberry Pi.

The [examples ](https://github.com/bnjmnp/pysoem/blob/master/examples/basic_example.py) is already ready to use to toggle a LED using an EL1259. I tried using this example and changing just the devices' product codes and the names in the SlaveSet.

However I don't get the LED to light up. Moreover, I don't think that I am fully understanding what the script does in particular.

1) Do I need the el1259_setup() function? If so, why do I need it? And what am I supposed to change in the .sdo_write()-calls?

`        slave = self._master.slaves[slave_pos]

        slave.sdo_write(0x8001, 2, struct.pack('B', 1))

        rx_map_obj = [0x1603,
                      0x1607,
                      0x160B,
                      0x160F,
                      0x1611,
                      0x1617,
                      0x161B,
                      0x161F,
                      0x1620,
                      0x1621,
                      0x1622,
                      0x1623,
                      0x1624,
                      0x1625,
                      0x1626,
                      0x1627]
        rx_map_obj_bytes = struct.pack(
            'Bx' + ''.join(['H' for i in range(len(rx_map_obj))]), len(rx_map_obj), *rx_map_obj)
        slave.sdo_write(0x1c12, 0, rx_map_obj_bytes, True)

        slave.dc_sync(1, 10000000)`

2) Do I toggle the LED via SDO or PDO? In the above mentioned example I found this code snippet, which suggests that I need PDOs:

`        toggle = True
        try:
            while 1:
                if toggle:
                    tmp[0] = 0x00
                else:
                    tmp[0] = 0x02
                self._master.slaves[2].output = bytes(tmp)

                toggle ^= True

                time.sleep(1)
`
But what are the tmp[0] = 0x00 and 0x02 supposed to mean?

Can someone please help me understanding this?

bnjmnp, 2022-05-16:
Within `.sdo_write(sdo_index, sub_index, setting, Boolean)`, the "setting" is supposed to be a list of bytes.
In the basic_example `slave.sdo_write(0x8001, 2, struct.pack('B', 1))` sets "Enable manual operation" for digital output 1 of the EL1259. Where `struct.pack('B', 1)` creates a list of bytes that is only 1 byte long, and the value of the byte is 1. Maybe the [pysoems readthedocs](https://pysoem.readthedocs.io/en/latest/coe_objects.html) can help you a little bit.
The `slave.sdo_write(0x1c12, 0, rx_map_obj_bytes, True)` updates the PDO mapping, for the EL1259. It defines what lays behind `self._master.slaves[2].output`.
If you are new to EtherCAT it might help, to first start with a GUI based EtherCAT master system.

bnjmnp, 2023-01-03:
Hi @MarcelloDavis, did you had any success with this project? Are you still working on it?

bnjmnp, 2024-02-24:
Closed due to inactivity of the OP. Please reopen the ticket if you need further help on this.

closed 2024-02-24 by bnjmnp