bnjmnp/pysoem #69: Trying to generate a build for Python 32 bits on Win32 but Missing file 'pysoem/pysoem.c' : No such file or directory

saftvendee, 2022-06-10:
Hi all
I need to generate a pysoem lib version for someone running Python 32 bits on his Win10 system (he's having Python compiled with MSC v.1900 32 bits INTEL). He can't use Python 64 bits due to his whole system install.

So I try to generate the right binairies from my machine (Win10 with MS build tools installed) typing `python .\setup.py bdist --plat-name=win32` (it this really the right thing to do ?) and I'm getting this:

```
> python .\setup.py bdist --plat-name=win32
running bdist
running bdist_dumb
running build
running build_py
creating build
creating build\lib.win-amd64-3.8
creating build\lib.win-amd64-3.8\pysoem
copying pysoem\__init__.py -> build\lib.win-amd64-3.8\pysoem
running build_ext
building 'pysoem.pysoem' extension
creating build\temp.win-amd64-3.8
creating build\temp.win-amd64-3.8\Release
creating build\temp.win-amd64-3.8\Release\pysoem
creating build\temp.win-amd64-3.8\Release\soem
creating build\temp.win-amd64-3.8\Release\soem\osal
creating build\temp.win-amd64-3.8\Release\soem\osal\win32
creating build\temp.win-amd64-3.8\Release\soem\oshw
creating build\temp.win-amd64-3.8\Release\soem\oshw\win32
creating build\temp.win-amd64-3.8\Release\soem\soem
C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DWIN32= -D_CRT_SECURE_NO_WARNINGS= -DEC_VER2= -I./pysoem -I.\soem\oshw\win32\wpcap\Include -I.\soem\oshw\win32 -I.\soem\osal\win32 -I.\soem\oshw -I.\soem\osal -I.\soem\soem -IC:\Python38\include -IC:\Python38\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt" /Tcpysoem/pysoem.c /Fobuild\temp.win-amd64-3.8\Release\pysoem/pysoem.obj
pysoem.c
c1: fatal error C1083: Impossible d'ouvrir le fichier source : 'pysoem/pysoem.c' : No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
```

Looks like `pysoem.c` file is missing. Can comeone please help me to solve this issue ?
Thanks in advance :)

bnjmnp, 2022-06-15:
Hi. Did you install [Cython](https://pypi.org/project/Cython/)?

Something you also need to change is the path to Packet.lib and wpcap.lib in the `setup.py`. On line 22 this
```py
    soem_lib_dirs = [os.path.join('.', 'soem', 'oshw', 'win32', 'wpcap', 'Lib', 'x64')]
```
should be replaced by this:
```py
    soem_lib_dirs = [os.path.join('.', 'soem', 'oshw', 'win32', 'wpcap', 'Lib')]
```

I guess it is sufficient to generate the bdist with an 32 bit Python distribution, the build already knows your target bit-width (so no `--plat-name=win32` needed).

Maybe there other things I'm missing, I never tried compiling for 32 bit.

emmanuelvendepro, 2022-06-15:
Thanks a lot for your reply @bnjmnp 😀
Indeed I don't have Cython installed 😅, I'll do it quickly and try to re-build with the fix you provide in the setup.py.
I'll keep you informed !