bnjmnp/pysoem #71: Setting SM & PDI watchdogs

alorcamo, 2022-06-22:
When communications takes more than the defualt 100ms the watchdog is triggered and changes the state from OP to Safe-OP. This switches off the outputs at potentially not safe moment. Worst case, switches output On/Off as communications changes from Ok to not Ok.

I need to switch it the SM & PDI watchdogs so whatever output state is remains. In SOEM there is a [Setting SM & PDI watchdogs](https://github.com/OpenEtherCATsociety/SOEM/issues/49#)](https://github.com/OpenEtherCATsociety/SOEM/issues/49) with this. 

Is there a way to do this with pysoem?

Thanks!

bnjmnp, 2022-06-22:
Hi,

At the moment the SOEM functions ec_FPRD()/ecx_FPRD() and ec_FPWR()/ecx_FPWR() are not exposed to the PySOEM API, so it is not possible to change the watchdog registers in ESC.

Maybe its time to add this somehow, to be able to do something like this:
```py
    watchdogDivRaw = slave.fprd(ECT_REG_WATCHDOG_DIV, 4)
    watchdogDiv  = 40 * (watchdogDivRaw + 2)
    watchdogPDORaw = pdoWatchdogTimeout / watchdogDiv
    slave.fpwr(ECT_REG_WATCHDOG_PDO_TIMEOUT, 4, watchdogPDORaw)
```
But it is not that easy because of the bytearray to integer conversion.

As as alternative we could add a `set_watchdog()` method to the slave class, where we have the logic of `setWatchDog` directly coded in PySOEM. This still keeps FPRD/FPWR not exposed, so its less flexible, but more simpler to use.
```py
    slave.set_watchdog(wd_time_ms=100)
```

What do you think?

Similar issue: I work on a new function inside PySOEM that enables users to change a slaves mailbox sizes. For the user it is then only one single function call.
```py
cdef class CdefSlave:
    ...
    def amend_mbx(self, mailbox, start_addr, size):
        if mailbox == 'out':
            self._ec_slave.SM[0].StartAddr = start_addr
            self._ec_slave.SM[0].SMlength = size
            self._ec_slave.mbx_wo = start_addr
            self._ec_slave.mbx_l = size
            cpysoem.ecx_FPWR(self._ecx_contextt.port, self._ec_slave.configadr, ECT_REG_SM0, sizeof(self._ec_slave.SM[0]), &self._ec_slave.SM[0], EC_TIMEOUTRET)
        elif mailbox == 'in':
            self._ec_slave.SM[1].StartAddr = start_addr
            self._ec_slave.SM[1].SMlength = size
            self._ec_slave.mbx_ro = start_addr
            self._ec_slave.mbx_rl = size
            cpysoem.ecx_FPWR(self._ecx_contextt.port, self._ec_slave.configadr, ECT_REG_SM1, sizeof(self._ec_slave.SM[1]), &self._ec_slave.SM[1], EC_TIMEOUTRET)
        else:
            raise AttributeError()
```
Here it makes even more sens to drag the code into PySOEM instead of exposing all involved SOEM struct elements and functions.

alorcamo, 2022-06-23:
Hi bnjmnp,

A method like `slave.set_watchdog(wd_time_ms=100)` would be an elegant addition to the pysoem capabilities. Probably a way to specify whether SM or PDI watchdog is needed as well. Either passing an enum or, perhaps more in pysoem style, `slave.set_sm_watchdog` and `slave.set_pdi_watchdog` methods.

Such methods make the code easy to use and read, including your proposed `amend_mbx`. The lower functions can be exposed as well for more specific use cases, advising simply that they are for power users / special cases.

bnjmnp, 2022-06-23:
Thank you for your opinion. The idea of power user functions is also good.
I don't have much time for the next few days, but I will tackle this topic in the next week(s).

alorcamo, 2022-06-23:
That sounds great, thanks!

bnjmnp, 2022-07-11:
Function [`set_watchdog()`](https://pysoem.readthedocs.io/en/latest/cdef_slave.html#pysoem.CdefSlave.set_watchdog) is now there to update the watchdog times.
Furthermore the (undocumented, protected) functions `_fprd()` and `_fpwr()` are available to read or modify any ESC register. You can look up how they are used in the [pysoem_register_test.py](https://github.com/bnjmnp/pysoem/blob/c869446e169ba063b1ff603f3f15f539ff9540c5/tests/pysoem_register_test.py) I just realized that the timeout_ms variable is named wrong in the fixture, it should be timeout_us.

alorcamo, 2022-07-22:
That's good news! I am looking forward to test it. Thanks @bnjmnp !

alorcamo, 2022-08-02:
I finally could test it and it works wonderfully. Thanks again for your support.

closed 2022-08-02 by alorcamo