bnjmnp/pysoem #74: Buffer underflow, processdata thread not reliable ?

Phiraos, 2022-07-12:
Hi,

This issue follows the #73.

I use a thread to periodicaly use processdata. 

But, even using a thread to send cyclicly processdata, I get systematicaly the error : PVT buffer underflow.

That means, somehow, the processdata loop is not reliable in terms of period (my loop period is approximately 4ms).

Here is the source :
```python
"""Toggles the state of a digital output on an EL1259.

Usage: python basic_example.py <adapter>

This example expects a physical slave layout according to
_expected_slave_layout, see below.
"""

import sys
import ctypes
import time
import threading
import math

from collections import namedtuple

import pysoem

class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('statusword', ctypes.c_uint16),
        ('position_actual_value', ctypes.c_int32),
        ('demand_position', ctypes.c_int32),
        ('demand_current', ctypes.c_int32),
        ('modes_of_operation_display', ctypes.c_int8),
        ('byte_padding', ctypes.c_int8),
    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('controlword', ctypes.c_uint16),
        ('target_position', ctypes.c_int32),
        ('target_velocity', ctypes.c_int32),
        ('modes_of_operation', ctypes.c_int8),
        ('byte_padding', ctypes.c_int8),
    ]

modes_of_operation = {
    'No mode': 0,
    'Profile position mode': 1,
    'Profile velocity mode': 3,
    'Homing mode': 6,
    'Cyclic synchronous position mode': 8,
    'Cyclic synchronous velocity mode': 9,
    'Cyclic synchronous torque mode': 10,
}

class LinmotActuator:

    BECKHOFF_VENDOR_ID = 0x960970
    EK1100_PRODUCT_CODE = 0x4c4e5449
    output_data = OutputPdo()
    output_data.modes_of_operation = modes_of_operation['Homing mode']
    check_thread_period = 0.002
    process_data_thread_period = 0.003

[... 202 lines cut ...]
                    LinmotActuator._check_slave(self._master.slaves[0], 0)
                if not self._master.do_check_state:
                    print('OK : all slaves resumed OPERATIONAL.')
            time.sleep(self.check_thread_period)

class LinmotActuatorError(Exception):
    def __init__(self, message):
        super(LinmotActuatorError, self).__init__(message)
        self.message = message

if __name__ == '__main__':

    print('ACMT v0 started')

    if len(sys.argv) > 1:
        try:
            LinmotActuator(sys.argv[1]).run()
        except LinmotActuatorError as expt:
            print('LinmotActuator failed: ' + expt.message)
            sys.exit(1)
    else:
        print('usage: LinmotActuator ifname')
        sys.exit(1)
```
I don't know how to make more stable the processdata loop.

bnjmnp, 2022-07-12:
For now I don't get where the error is coming from. Dose it pop up on the display of the acutator?
Dose it happen in the CSP mode only or also in homing?

bnjmnp, 2022-07-12:
Dose the actuator go into SafeOP state when the error occurs, and stops the motor?

With Python and non-realtime operating systems you will always have big jitter, also network interface cards connected via USB are not good to achieve realtime. Your actuator seems not to like that jitter. I thought "Profile Position" mode would be more resistant to jitter, but this seems also not be the case for your actuator?

Phiraos, 2022-07-13:
> For now I don't get where the error is coming from. Dose it pop up on the display of the acutator? Dose it happen in the CSP mode only or also in homing?

I get the error on the utilitary software of the vendor. I use it for monitoring. I get the error only in CSP mode.

Phiraos, 2022-07-13:
> Dose the actuator go into SafeOP state when the error occurs, and stops the motor?
>
> With Python and non-realtime operating systems you will always have big jitter, also network interface cards [...]

When the error occurs, the motor stops working but I can't tell is state. 
I use a ethernet port on a NUC.
I don't know how the send_processdata works but, I was thinking maybe the update loop and the processdata loop try to access in the same time the output and that's why it gets out of period.