bnjmnp/pysoem #75: No connection to EtherCat slave

zlymudzyn, 2022-07-13:
Hi,

I have quite an odd problem: I am trying to connect to a slave via EtherCAT using pysoem. When I am directly connected to the device, everything works fine. Problem occurs when I am trying to connect to it via Netgear managed switch. Then, the device is not visible at all. I've tried using a program Codesys, which is also able to simulate an EtherCAT master and, to my surprise, it is able to find the slave and connection works fine.

Using pysoem I am using the minimal_example.py file and it does not find anything. I am running it on Raspbian.

What could be causing the problem here? I would rather use pysoem than Codesys, as it gives me way more possibilities.

bnjmnp, 2022-07-14:
As far as I know managed switches should not be put into an EtherCAT network. Almost all EtherCAT devices have an in and out port, so you connect them one after the other ([https://www.ethercat.org/en/technology.html#1.3](https://www.ethercat.org/en/technology.html#1.3). If your devices only have on port you need an "EtherCAT switch" like the [Beckhoff CU1128](https://www.beckhoff.com/en-en/products/i-o/infrastructure-components/cuxxxx-epxxxx-ethercat-components/cu1128.html).

zlymudzyn, 2022-07-14:
The problem is that the device I am testing has multiple protocols that it can communicate through: Profinet, Ethernet IP, Ethercat. Therefore I would rather not use a dedicated EtherCAT switch as it would mean I would need a separate test bench for testing EtherCAT only. Doable, just not preferable.

The device I am testing has one communication port for all these protocols. There is a switch inside of it.

From what I can understand EtherCAT can be used with with this managed switch, as Codesys finds the device as EtherCAT slave no problem(I have created a separate VLAN for my raspberry and the device), but using pysoem it does not find the device and I find it hard to understand why that might happen.

Using unmanaged switch it also works, but due to some limiting factors I cannot use it in my project.
Any ideas?

bnjmnp, 2022-07-14:
Not really. You could use Wireshark to see what Codesys is doing different. You may also go through the [issues of the SOEM project](https://github.com/OpenEtherCATsociety/SOEM/issues?q=is%3Aissue) itself, this is what PySOEM is based on. With a bit of luck you will find there the information you are looking for. Or ask for help there.