bnjmnp/pysoem #79: DC SYNC and PDO Loop?

tom9672, 2022-09-05:
DC SYNC and PDO Loop, is it okay to choose only one of the two? 
If the master station uses the PDO sending and receiving cycle, it does not need to configure dc_sync? 
In other words, can I not need to send and receive processdata after using dc_sync?

bnjmnp, 2022-09-06:
Yes, for CSP mode DC synchronization might be required, it makes sens there. But it could be that on drives from another vendor DC synchronization might not be required.
With Python/PySOEM exact DC synchronization is not possible, because we cannot achive realtime.

tom9672, 2022-09-07:
I'm trying to get the motor to run at variable velocity , and the positioning is precise when changing velocity.
[image]
I tried to use PP mode with Instantly update and PV mode. it can achieve continuous speed change without dropping zero. but the position is not very accurate, the reason: for example, I want change velocity from 100 to 200, while position is 10000. In pp Instantly update and PV, I have to check if the current position is at a range of target postion. I using ``` current_position > 10000``` or ```current_position in range(9900,11000)```, which make the position of changing velocity not precise.

PP mode with not immediately updated, the position of changing velocity is precise but it will make velocity ​​decelerates to zero before accelerating to target speed.

Is there a way to make the make the target position of changing velocity as precise as much.

tom9672, 2022-09-09:
I changed a window PC to a raspberry pi on linux with rt-kernel and one cpu isolation. Now I can reduce PDO loop time gap, and not get incorrect wkc.

I continue to reduce the time sleep of PDO loop and position monitor loop. While I set the time sleep of this two loop to 0.00000001 and smaller, each adjacent position monitoring feedback value point has more and more the same value, that means the PDO loop can’t get the current position at such small time sleep gap, right?

At this situation, I run the motor at 3000rpm, the biggest position gap is around 200 to 300(10000=1round). This is not good enough…

So, it seems means, at free run mode with PP or PV mode, even the PDO loop is quick, but maybe some other reasons, still unable to monitor the real time position feedback.

tom9672, 2022-09-13:
Find the reason, cause the device I’m using support the smallest PDO time gap is 250 μs.

GARBAGE-WALL-E, 2022-10-17:
@tom9672 I am walking on the somewhat similar path as you did, and facing the similar issues, Can you please help me out, I would really appreciate it. (Link to the issue [https://github.com/bnjmnp/pysoem/issues/84#issue-1411168250](url) ) Did you get the non jumpy motion in CSP mode ? How?

closed 2022-10-24 by tom9672