bnjmnp/pysoem #80: Is EOE not implemented or am i initializing incorrectly ! [proposed Label] help wanted

rohithsantosh13, 2022-09-14:
Hi i am using Twin CAT to configure the EtherCAT master to use EOE , but when in run the Pysoem script to read the data, the EOE-configurations made from TWINCAT are being removed. can anyone suggest a solution or workaround for this issue

bnjmnp, 2022-09-15:
Currently EoE is not supported in any way by PySOEM, and it is not possible to run TwinCAT and PySOEM in parallel using the same network adapter. Event if the EoE related SOEM functions would be exposed by PySOEM, rerouting the Ethernet to a virtual Windows network adapter, like TwinCAT is doing this, is probably not an easy task.

rohithsantosh13, 2022-09-29:
@bnjmnp Thanks for the reply. are there any plans to support EoE on Pysoem ?

bnjmnp, 2022-10-03:
It is not planned yet.
I cannot oversea the effort, and I don't have a good idea on how to test it.
Again, don't expect a virtual Ethernet port in your OS, you will only get something like a read and write function where you need route your Ethernet frames to.