bnjmnp/pysoem #81: Can't decode slave information

MarcelloDavis, 2022-10-04:
Hello,

i was just trying to simply read out the slave device's name via sdo_read. The lines of code I used are:

```
import sys
import pysoem
import struct

master = pysoem.Master()
master.open("eth1")

EL1809 = master.slaves[1]
device_name = EL1809.sdo_read(0x1008, 0).decode('utf-8')
```

However, when I ran these lines of code, the output says: 'utf-8' codec can't decode byte 0x80 in position 0: invalid start byte
Do you have any guesses as to what might be the problem?

bnjmnp, 2022-10-12:
Hi @MarcelloDavis,

Did you try to call `master.config_init()` prior to accessing the device? This is required to establish any communication.

s2cihero, 2022-11-17:
Just try sleep() func. after  "EL1809 = master.slaves[1]" after this. Cause if u open ethercat com,just register is going out.

for this solution.sleep() function.3 sec is enough i tried.

SukritMachineTools, 2023-02-16:
try this:

`import pysoem
import ctypes
master = pysoem.Master()

master.open('\\Device\\NPF_{43350287-BA24-4787-A337-A02C7CE4AEA2}')

if master.config_init() > 0:
    for device in master.slaves:
        print(f'Found Device {device.name}')
        pav=device.sdo_read(0x1008, 0).decode('unicode_escape')
        pav2 = ctypes.c_uint32.from_buffer_copy(device.sdo_read(index=0x6064, subindex=0x00)).value
        print(pav)
        print(pav2)`

replace index with register address you want to read