bnjmnp/pysoem #84: Can not control Velocity in CSP Mode

GARBAGE-WALL-E, 2022-10-17:
@tom9672 I am walking on the somewhat similar path as you did, and facing the similar issues, Can you please help me out, I would really appreciate it. (Apologies for butchering your code)  DId you get the non jumpy motion in CSP mode ? How?

The issue is some how in CounterClockWise(+ve direction) rotation the servo moves very smoothly, But with the same code when I do the decrement of the position ClockWise(-ve) it gives very Jumpy motion. 
Tried Changing DC Sync Time but it affects only the smooth motion (makes faster or slower), But the ClockWIse rotation issue remains.
I have tried @tom9672 's Idea of writing velocity by sdo but it doesn't work as well. I assume x6081 is not effective in CSP mode it is for PP mode, I have tried TwinCat and it Controls Velocity by x60FF, (BTW servo moves perfectly by twin can in CSP mode)

Apologies for messy code it's just for reference, nothing much to look in to 
```
class InputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [  
        ('Errorcode', ctypes.c_uint16), ###x603F0010 
        ('status_word', ctypes.c_uint16),  ###x60410010 actual_position
        ('actual_position', ctypes.c_int32),  ### 60640020 velocity
        ('TorqueActualValue', ctypes.c_int16),  ###x60770010
        ('Modesofoperationdisplay', ctypes.c_int8),  ##x60610008
        ('Touchprobestatus', ctypes.c_uint16),  ###>#x60B90010
        ('Touchprobepos1posvalue', ctypes.c_int32),  ###>#x60BA0020
        ('Touchprobepos2posvalue', ctypes.c_int32),  ###>#x60BC0020
        ('Digitalinputs', ctypes.c_uint32),  ###>#x60FD0020

    ]

class OutputPdo(ctypes.Structure):
    _pack_ = 1
    _fields_ = [
        ('Control_Word', ctypes.c_uint16),  ###0x6040 
        ('Target_Position', ctypes.c_int32),  ###0x607A  
        ('Velocity', ctypes.c_int32),  ###0x60FF 
        ('TargetTorque', ctypes.c_int16), ###60710010    
        ('Mode_Operation', ctypes.c_int8), ###x60600008  
        ('TouchProbeFunction', ctypes.c_uint16), ###x60B80010 
        ('MaxProfileVelocity', ctypes.c_uint32), ###x607F0020 

    ]

class MachineWork:
    def __init__(self,port):
        self._port = port
        self._master = pysoem.Master()
        self._master.in_op = False
        self._master.do_check_state = False
        self._pd_thread_stop_event = threading.Event()
        self._ch_thread_stop_event = threading.Event()
        self._pos_thread_stop_event = threading.Event()
        self._mov_thread_stop_event = threading.Event()

    def _device_config_func(self,slave_pos):
        device = self._master.slaves[slave_pos]

        device.sdo_write(0x2005,0x1,bytes(ctypes.c_uint16(1))) ###RESET ERROR
        
 
        device.sdo_write(0x1C12,0x0,bytes(ctypes.c_uint8(0)))
        device.sdo_write(0x1C13,0x0,bytes(ctypes.c_uint8(0)))
        device.sdo_write(0x1600,0x0,bytes(ctypes.c_uint8(0)))

        device.sdo_write(0x1A00,0x0,bytes(ctypes.c_uint8(0)))

        device.sdo_write(0x1600,0x1,bytes(ctypes.c_uint32(1614807056)))#6040 c_word
        device.sdo_write(0x1600,0x2,bytes(ctypes.c_uint32(1616904200)))#6060 mode
        device.sdo_write(0x1600,0x3,bytes(ctypes.c_uint32(1618608160)))#607A t_pos
        device.sdo_write(0x1600,0x4,bytes(ctypes.c_uint32(1619066912)))#6081 t_speeed PP
        ##device.sdo_write(0x1600,0x5,bytes(ctypes.c_uint32(1627324448)))#60FF t_speeed PV
        

        device.sdo_write(0x1A00,0x1,bytes(ctypes.c_uint32(1614872592)))#6041 s_word
        device.sdo_write(0x1A00,0x2,bytes(ctypes.c_uint32(1617166368)))#6064 pos
        device.sdo_write(0x1A00,0x3,bytes(ctypes.c_uint32(1617690656)))#606C speed

[... 242 lines cut ...]
                        'MESSAGE : slave {} recovered'.format(pos))
            else:
                slave.is_lost = False
                print('MESSAGE : slave {} found'.format(pos))

class HandleError(Exception):
    def __init__(self, message):
        super(HandleError, self).__init__(message)
        self.message = message

def xox2():
  def threadxox2():
    while True:
          try:
                time.sleep(0.1)
          except:
            print("pos update except")
            time.sleep(0.5)
       
  t = threading.Thread(target=threadxox2)
  t.start()
port = '\\Device\\NPF_{23FFFBFE-C8A4-4C57-8FF3-65AD208FA513}'
machine = MachineWork(port)
machine.setup()
```

tom9672, 2022-10-20:
As the handbooks of ethercat servo drivers. The velocity of CSP mode is decided by servo driver itself (according to snchronization time and target position), user unable to set target velocity in this mode. If you need to let the motor runs at a targer velocity, I guess you'd better use PV or CSP mode.

GARBAGE-WALL-E, 2022-10-20:
@tom9672 Thank you so much for the reply.
  I had doubts because, I went through the datasheet of several brand drives and one of them mentioned 70FF affects the CSP mode, also the TwinCat writes the velocity to 70FF in CSP.

  But anyways, I wrote the Acc and Dcc algorithms based on position and they are working fine, Also Increased the sync time and the directional motion issue is under the rug now.

> Yet it really bothers me, Why motion is smooth in one direction(+Ve) and not in other direction (-Ve)? You can notice it considerably as you lower down sync time.
  I have looked in to DemandedPosition value( at not very high refresh rate though) and it seems ok. Also I have tried different brand servo and the issue is still the same.
60B1, 60B2 are also set to zero.
Do you think, Does it have anything to do with underlying pysoem or soem?
Am I missing something? Any parameter?

closed 2022-10-22 by GARBAGE-WALL-E