bnjmnp/pysoem #85: The performance of SOEM and PYSOEM?

tom9672, 2022-10-20:
I almost mastered using pysoem to control servo motors now.  C is known to be more efficient than python, Am I thinking that SOEM will have much higher performance than pysoem? Is it worth learning SOEM. 

In addition to the performance of the programming language, in order to achieve higher system real-time performance, my approach is to use Linux real-time kernels and isolated cores. Is there any other way to improve performance, or is there a way to verify that real-time performance is good?

bnjmnp, 2022-10-23:
It is very likely that you will have a much better performance using C and SOEM directly.
If you are familiar with C, learning SOEM is not that difficult, PySOEM and SOEM functions are somewhat similar, just the naming is sometimes slightly different. You can always check the wrapper code inside [pysoem/pysoem/pysoem.pyx](https://github.com/bnjmnp/pysoem/blob/c869446e169ba063b1ff603f3f15f539ff9540c5/pysoem/pysoem.pyx). For example `ecx_config_init()` in SOEM is `config_init()` in PySOEM, `ecx_config_map_group()` in SOEM is `config_map()` in PySOEM. One day I might write a migration guide.
With a bit of luck you may get a little performance boost with Python 3.11.
Learning C is anyway a good idea if you like to go deeper into embedded software.

I don't know any other options to improve the performance.
Although SM synchronization might be good enough four your application, note that for correct DC synchronization you need to synchronize the master systems clock with the device clocks in the EtherCAT network like in this SOEM example [SOEM/test/linux/ebox/ebox.c](https://github.com/OpenEtherCATsociety/SOEM/blob/6a5babfec9d1c6c892ac6fededfec7a508ee2320/test/linux/ebox/ebox.c). This is probably not doable with Python/PySOEM.

To verify real-time performance and correctness you need to ether record the traffic on the network with an Ethernet sniffer device. Or you have a development board for an EtherCAT device, where you write your own firmware that somehow can verify timings in a simpler way.

tom9672, 2022-10-24:
Thank you for your detailed reply!  👍

closed 2022-10-24 by tom9672