bnjmnp/pysoem #86: Failed to download pysoem using pip in the win10 system. Urgent!

EventPath, 2022-11-08:
I have installed Npcap (compatible with Winpcap), but the error prompt when using pip to download pysoem is: LINK: fatal error LNK1181: cannot open input file 'wpcap. lib'

EventPath, 2022-11-08:
I have solved the problem. You only need to place three lib files: libsoem.lib, Packet.lib, wpcap.lib in the file path: C: Program Files (x86) Microsoft Visual Studio 14.0 VC lib

bnjmnp, 2022-11-18:
Are you using a 32-bit Python distribution?

Tyhmg, 2022-11-22:
Yes, the python I use is 32-bit.

bnjmnp, 2022-11-22:
Someone else created an issue in regards to 32-bit Python, maybe you will find an alternative solution there: [#69](https://github.com/bnjmnp/pysoem/issues/69).

I'm wondering why some people need to use a 32-bit Python? Maybe it is a good idea for me to update the build to also work for 32-bit Python right away?