bnjmnp/pysoem #89: Synapticon Node2000

RasitEvduzen, 2022-11-14:
Hello everyone,

I'm trying to control Synapticon Node2000 using pysoem. I think ı don't understand clearly how to  EtherCAT communication works. I'm try several code and read Synapticon any information.But I don't know, how to control node 2000 driver using pysoem and how to initialization EtherCAT communication protocol. Could anyone help me?

Regards.

bnjmnp, 2022-11-18:
It is difficult to give advice to people new to the EtherCAT technology.
Do you have anything specific where you stuck?

In some issues, there are code examples on how operate a servo drive that is based on DS402 drive profile:
* [#30](https://github.com/bnjmnp/pysoem/issues/30)
* [#33](https://github.com/bnjmnp/pysoem/issues/33)
* [#43](https://github.com/bnjmnp/pysoem/issues/43)
* [#70](https://github.com/bnjmnp/pysoem/issues/70)
* [#73](https://github.com/bnjmnp/pysoem/issues/73)
* [#76](https://github.com/bnjmnp/pysoem/issues/76)

RasitEvduzen, 2022-11-18:
@bnjmnp Thank you for your advice. I'll read asap.