bnjmnp/pysoem #90: CIA402 ControlWord Read/Write all bit's.

s2cihero, 2022-11-17:
Hello friends.

Just i m trying servo motor enable on cia402.and this com. i need send and write some bits(x6040-Controlword) but if i try to read sdo_read,just only giving first bit.i need to read all bit.i have ABB MicroFlex e190 servo drive.

can anyone help me ?

bnjmnp, 2022-11-17:
Do you (SDO)read the Controlword or the Statusword? I think reading the Controlword is not necessary.

Michaelicious, 2023-11-09:
To read status you can read statusword 0x6041, after decoding with ctypes it returns whole int for me. 
you can try writing the int that translates to the binary you need, translate to binary using ctypes.
here's my functions for enable, I have a custom `readSDO()` that decodes in the background using ctypes.c_uint16 for this specific object. 

```def clearFaults(self):
        self.writeSDO(0x6040, 0, 128)            # Clear faults
        statusword = self.readSDO(0x6041, 0) 
        print(f'Cleared faults: 0x6040,0,value=128.\nstatusword 0x6041:\t{statusword}\n')
        # if statisword is something: return False
        # else: retrun True

    def enableSlave(self):
        print(f'Initial statusword:\n {self.readSDO(0x6041, 0)}\n')
        self.clearFaults()
        self.writeSDO(0x6040, 0, 6)
        print(f'Wrote 6. statusword:\n{self.readSDO(0x6041, 0)}\n')
        self.writeSDO(0x6040, 0, 7)
        print(f'Wrote 7. statusword:\n{self.readSDO(0x6041, 0)}\n')
        self.writeSDO(0x6040, 0, 15)
        print(f'Wrote 15. statusword:\n{self.readSDO(0x6041, 0)}\n')```

more of my code at issue #121 , Hope it helps.