bnjmnp/pysoem #95: Multiple masters on same computer using same working counter

hstroetgen, 2023-01-17:
I use pysoem for an end-of-line test system. This tester uses one master to control Beckhoff terminals with digital/analog IO and the second master controls the DuT (DUT, a servo drive ).
When I run my test cases (two instances of my master are created), I can observe that the working counter of the master controlling the Beckhoff terminals becomes wrong as soon as something happens to the DuT (power cycle, state change, etc.).
I was able to solve the problem by installing my master application with a different name in parallel to the original master in my virtual environment (ugly solution).
Is it possible to fix this? Is this related to pysoem at all?

Translated with www.DeepL.com/Translator (free version)

bnjmnp, 2023-01-17:
Hi @hstroetgen, I saw that coming one day. The check if multiple instance of the master class are working properly was on my todo list, with the hope it would work right away, or this would barely be needed by users, but I lived with the fear that it doesn't work.
The SOEM library is supporting it, so this should not be the limiting factor. I will look into it the coming weeks, I planned to further work on PySOEM anyway.

hstroetgen, 2023-01-18:
Hey @bnjmnp, thank you very much! That's great news.

bnjmnp, 2023-02-12:
I did some tests on this, but until now I could not see any issues when using 2 network interface cards at the same time.
One time I basically created a Python script similar to "basic_example.py", where I use two instances of the "BasicExample" class and then let both run functions run in a tread. That seemed to work fine.
From my experience, it sometimes helps to increase the timeout argument given to `receive_processdata()` when having issues with the process-data working counter.
Anyway I will try to further investigate the multi master use case.