bnjmnp/pysoem #96: can not searching slaves...

Mrkeyn, 2023-01-19:
Hello, everyone. nice to meet you 
 I need some help.
my pc can not searching slaves...
os : windows 10
python : 3.11
pysoem : 1.0.0.7
Slave devices : fastech - Ezi-IO EtherCAT INPUT16

-------------------------------------------------------------------------------------
import sys
import pysoem

def get_adaptername():
    adapters = pysoem.find_adapters()
    for i, adapter in enumerate(adapters):
        if(adapter.desc == 'TwinCAT-Intel PCI Ethernet Adapter (Gigabit)'):
            adpt = adapter.name
    return adpt

try:
    master = pysoem.Master()
    log(0, '<< master open started >>')
    master.open(get_adaptername())
except Exception as e:
    master.close()
    log(0, e)

log(0, 'searching slave start')
if master.config_init() > 0:
    device_foo = master.slaves[0]
    device_bar = master.slaves[1]
else:
    log(0, 'no device found')

master.close()

================================================================
17:52:10,205 - root [DEBUG] : searching slave start
17:52:10,206 - root [DEBUG] : no device found

================================================================

<EtherCATInfo xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="EtherCATInfo.xsd" Version="1.11">
	<Vendor>
		<Id>#x0FA00000</Id>
		<Name>FASTECH</Name>
~~~~
		<Devices>
			<!--Ezi-IO EtherCAT DIO--><Device Physics="YY">
				<Type ProductCode="#x02002" RevisionNo="#x00001">Ezi-IO EtherCAT INPUT16</Type>
				<Name>Ezi-IO EtherCAT INPUT16</Name>
				<GroupType>Ezi-IO Controller</GroupType>
				<Fmmu>Inputs</Fmmu>
				<Sm DefaultSize="2" StartAddress="#x1000" ControlByte="#x00" Enable="1">Inputs</Sm>
				<TxPdo Fixed="1" Mandatory="1" Sm="0">

bnjmnp, 2023-01-21:
Hi. That no device was found could be an issue with access rights, not sure. I never onderstood why on Windows sometimes one needs administrator rights.

Mrkeyn, 2023-02-15:
> Hi. That no device was found could be an issue with access rights, not sure. I never onderstood why on Windows sometimes one needs administrator rights.

thanks. i solved on ubuntu : )