bnjmnp/pysoem #97: config_map() SdoError: Subindex does not exist

jvbpv, 2023-02-03:
Hi! 

I have used this pysoem library to interface with various ethercat devices with success. 
Now I have run into issue attempting to interface with a ethercat turbo pump.

I am attempting to access PDO input and output data. 
According to the device ethercat manual, the 0x1600 and the 0x1A00 addresses contain the defualt RxPdo and TxPdo mappings respectively. There is 1 byte to access in the process output data and 7 bytes in the input data.

The following line causes issues:
```ecat.master.config_map()```

I get the following error:
[image]

I have tried looking for a solution browsing through the issues on this repository but I have not figured out a possible solution.

Is there an underlying understanding to the config_map() function that I am missing?

I greatly appreciate any help on leading me in the right direction to solve this. Thank you

bnjmnp, 2023-02-05:
Hi.
`config_map()`, calls if available the `master.slaves[x].config_func` of each device attached, then reads out the PDO mapping and PDO configuration of all devices and finally changes the state from PreOP to SafeOP.
The device you are using seems to have issues when reading out the default PDO mapping defined by object 0x1C12 (and 0x1C13).
You see the errors are caused by SOEM reading object 0x1C12 (7186 decimal) subindex 65 to 128.
I guess that subindex 0 of object 0x1C12 is set to 128 for some reason.
You could try to configure the entries of object 0x1C12 to the desired/default mapping.
At best with the `master.slaves[x].config_func`, like in the `minimal_example.py` done with the `el3002_setup()` function.

jvbpv, 2023-02-10:
Hello,

Thank you for your response!
I have given it a few attempts with your suggestion and many other attempts for setting the right configurations. I created a setup function and set it to `master.slaves[x].config_func` for when `config_map` is called.
I have attempted to configure the default PDO mappings and my own user set PDO mappings to 0x1c12 and 0x1c13 and both have failed to work.

This is the error I run into:
[image]

This is the error I get when after my setup function is completed and the master attempts to set the device to SAFE_OP. I get an invalid configuration error LED flashing on the pump.

I use the SOEM slaveinfo program to read the default set PDO mappings afterwards and this is what I see:

``` .log
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s8 succeeded.
Time:1675990481.612 SDO slave:1 index:1c12.00 error:08000021 Data cannot be transferred or stored to the application because of local control
Time:1675990481.617 SDO slave:1 index:1c13.00 error:08000021 Data cannot be transferred or stored to the application because of local control
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:netX
 Output size: 3bits
 Input size: 56bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000005e9 ID: 10000000 Rev: 01010002
 SM0 A:1000 L: 128 F:00010036 Type:1
 SM1 A:1080 L: 128 F:00010032 Type:2
 SM2 A:1100 L:   1 F:00010074 Type:3
 SM3 A:1d00 L:   7 F:00010030 Type:4
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:2 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000001 Ll:   7 Lsb:0 Leb:7 Ps:1d00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      
  [0x0000.1] 0x7000:0x02 0x01 BOOLEAN      
  [0x0000.2] 0x7000:0x03 0x01 BOOLEAN      
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0001.0] 0x6000:0x01 0x01 BOOLEAN      
  [0x0001.1] 0x6000:0x02 0x01 BOOLEAN      
  [0x0001.2] 0x6000:0x03 0x01 BOOLEAN      
  [0x0001.3] 0x6000:0x04 0x01 BOOLEAN      
  [0x0001.4] 0x6000:0x05 0x01 BOOLEAN      
  [0x0001.5] 0x6000:0x07 0x01 BOOLEAN      
  [0x0001.6] 0x6000:0x09 0x01 BOOLEAN      
  [0x0001.7] 0x6000:0x0A 0x01 BOOLEAN      
  [0x0002.0] 0x6000:0x11 0x10 INTEGER16    
  [0x0004.0] 0x6000:0x12 0x10 INTEGER16    
  [0x0006.0] 0x6000:0x13 0x10 INTEGER16    
End slaveinfo, close socket
End program
```

Before any configurations I am able to see that 0x1c12 subindex 0 contains 128 like you mentioned as well as values 1, and 0xb981:

```.log 
    0x00      "Number of elements"                          [UNSIGNED8        RWR_R_]      0x80 / 128
    0x01      ""                                            [UNSIGNED16       RWR_R_]      0x0001 / 1
    0x02      ""                                            [UNSIGNED16       RWR_R_]      0xb981 / 47489
```

After making configuration changes this is what I get:
```.log 
0x1c12      "Sync Manager 2 PDO Assignment"               [ARRAY  maxsub(0x03 / 3)]
- Time:       0.000 SDO slave:1 index:1c12.03 error:06090011 Subindex does not exist

- 
    0x00      "Number of elements"                          [UNSIGNED8        RWR_R_]      0x01 / 1
    0x01      ""                                            [UNSIGNED16       RWR_R_]      0x1600 / 5632
```

I followed the turbo pump Ethercat manual for setting user PDOs and I get the same error.
Any suggestions of what the issue could be?