bnjmnp/pysoem #98: Cann't connect with network adapters ID

SukritMachineTools, 2023-02-08:
`import pysoem

master = pysoem.Master()

master.open('Your network adapters ID')

if master.config_init() > 0:
    for device in master.slaves:
        print(f'Found Device {device.name}')
else:
    print('no device found')

master.close()`

Shows error :
**ConnectionError: could not open interface 00-E0-4C-16-C4-58**

I had replaced **network adapters ID** with every **mac address and ip address** of my PC (in **Window 10 & Linuxcnc**), but Things doesn't work.

**Please guide me how to use this code in my application ??**

bnjmnp, 2023-02-09:
Hi. you could simply use the [find_adapters.py](https://github.com/bnjmnp/pysoem/blob/9dc522808ee2587fb3586e78128e3498d14c8df2/examples/find_adapters.py) script to print out all available IDs and then copy the right one into you code.

SukritMachineTools, 2023-02-10:
Your method works in **Windows 10** but It doesn't work in **Raspberry Pi Os** and **Linuxcnc os**.

In that os, by find_adapters.py it show **eth0** as adapter name but when i put it in place of **Your network adapters ID**. It doesn't work . 

there it shows error:
**`ConnectError :  could not open interface eth0`** .

bnjmnp, 2023-02-10:
It is not nice but did you try running you Python code with administrator privileges? Could be that it will work then. SOEM uses lower level OS drivers and Linux limits the access to this stuff.

SukritMachineTools, 2023-02-11:
Yes I have given excitable privileges to Python code. It can execute  **python files and PymodbusTcp modules** easily. But It show error when use **pysoem**.  

there it shows error:
**`ConnectError :  could not open interface eth0 .`**