bnjmnp/pysoem #102: EL4004 Analog Output Problem: Attempt to write to a read only object

WeLikeIke, 2023-03-24:
Hello,
Thank you very much for this great library, by reading issues here on github and playing around with the examples I was able to create some scripts that correctly manage Digital I/O and Analog Input...

Regarding Analog Outputs I am stumped, since I'm at my wits end I would like to ask some help to troubleshoot:

**What happens?**
In a configuration connecting EK1100 (Main terminal), EL1018 (Digital In), EL2004 (Digital Out), EL3008 (Analog Input) and EL4004 (Analog Output) I am able to reach `OP` state with all of them (tested by checking `slave.state` and `slave.read_state()`), I am able to read the `slave.inputs` bytes of the EL1018 and EL3008 and write the `slave.outputs` (Digital Outputs) of the EL2004.
However I am NOT able to write the Analog Outputs of the EL4004.

When I try I receive the following console print:
`(2, 7186, 0, 100728834, 'Attempt to write to a read only object')`

**Assumption I made**
It looks to me that any Beckhoff module that is correctly connected will launch with its own default SDOs and as such would be immediately operational EVEN WITHOUT setting `slave.config_func`.
This assumption seems to hold because, even if the 2 example scripts `basic_example.py` and `minimal_example.py` set some PDOs for an EL1259 (Digital Input & Output) and EL3002 (Analog Input), I am able to govern an equivalent I/O WITHOUT setting `slave.config_func`.

**What did I try?**
Starting from the idea that the above assumption was correct, I suspected that the EL4004 that I was using was faulty, afterall it was telling the master that it was in `OP` state but could not be operated.
That was, unfortunately not the case, I had another couple of EL4004 that I could play with and nothing changed.

Then I thought that it might be a problem with the EL4004 in general, but upon swapping it with another Analog Output module, the EL4182, the same behaviour presented itself.

At this point I am of the opinion that the above assumption is incorrect, as such I started suspecting that the EL4004, or maybe Analog outputs in general, would require the use of a `slave.config_func` to properly enter operational mode.
But no matter what I tried, every meaningful configuration would require the use of `slave.sdo_write()`, which will give the usual `(2, 7186, 0, 100728834, 'Attempt to write to a read only object')` error.

I don't have a TwinCat license from Beckhoff but I was able to find a "ethercat circuit debugger" that would let me manually read the inputs and write the outputs of an EtherCat circuit.
This completely excluded the possibility of a faulty I/O module, since using this debugger I was able to write into the Analog Outputs from its GUI.

I was even able to contact the man that created the above mentioned program and he concluded that it might be a problem with the initial handshake that the master executes to know in which order the slaves are configured in the EtherCat circuit.

If his hunch is correct, it might be a problem with the library since I do not directly deal with the starting handshake myself.

**Minimal reproducible example**
```import sys
import time
import pysoem

# CONSTANTS
# Chosen Ethernet network adapter (through trial and error)
ETH = "Realtek PCIe GbE Family Controller"
# Known Id for Beckhoff products in the EtherCAT protocol
BECKHOFF_VENDOR_ID = 2
# EtherCAT states
INIT   = pysoem.INIT_STATE
PREOP  = pysoem.PREOP_STATE
SAFEOP = pysoem.SAFEOP_STATE
OP     = pysoem.OP_STATE
# Analog resolution
INT_TO_VOLT = 10.0 / 2 ** 15

# Function to correctly return the network adapter associated with the current laptop's rightside Ethernet port
def just_connect():
    adapters = pysoem.find_adapters()
    for a in adapters:
        if a.desc == ETH:
            return a.name

    raise ConnectionError("Could not find the chosen Network adapter ({})".format(ETH))

# Function to display the name and number of I/O bytes that a slave has available
# Bytes are not equal to I/O in general
def io_print(slave: pysoem.pysoem.CdefSlave):
    print("{} I/O: {}/{}".format(slave.name, len(slave.input), len(slave.output)))

# Function to transform given bytes into a list containing which Digital Inputs are on
def bytes_to_Din(b: bytes, n_bytes: int, n_Din: int):
    if len(b) != n_bytes:
        raise Exception("Mismatch between the number of given bytes ({}) and the number of expected bytes ({})".format(len(b), n_bytes))

    v = int(b.hex(), 16)
[... 238 lines cut ...]
Maybe "minimal" is incorrect because I use many utility functions but I think there should be no problems here.

**To reproduce**
Setup a EtherCat circuit as follows: EK1100 - EL4004 - EL3008

In the script above substitute the string constant `ETH` with the name of your Network Interface (on Windows it works, I don't know on Linux, you might need to rewrite the initial connection of the master) 

Running the script above without arguments will print all the times when there is a writable SDO but it cannot be written because of the usual `(2, 7186, 0, 100728834, 'Attempt to write to a read only object')` error, it also shows that I have NO issue using a similar configuration function to setup the EL3008.

Running the script with any argument, doesn't matter what, would trigger the usage of the `EL4004_setup` function instead of the `EL4004_safe_setup` function as the `master.slaves[1].config_func`, showing the usual error in the console.

**Additional info**
This is the page relative to the Error given by the Beckhoff module:
https://infosys.beckhoff.com/english.php?content=../content/1033/ethercatsystem/1037010571.html&id=

This is the page relative to the EL4004, which gives some information regarding SDOs, but again I cannot write to any of them:
https://infosys.beckhoff.com/english.php?content=../content/1033/el6695/1317558667.html&id=

And here is the display of the EL4004, it is written that when the module is in the `OP` state the leds would be all green, but my EL4004 (even if the `slave.state` tells me that it is in `OP`) is continuously blinking which would indicate (if this is to be trusted) that it is still in `PREOP`:
https://infosys.beckhoff.com/english.php?content=../content/1033/ep9576-1032/9905045643.html&id=

Let me know if there is any imformation that I can provide, 
I would be REALLY glad if you could help me troubleshoot this, I have been wasting 3 days on it...

Thanks for your time

bnjmnp, 2023-04-01:
Looks like the EL4004 has fixed PDOs, thus there is no config_func needed. This is why you get the SDO error (Attempt to write ..).

I think the LEDs on the EL4004 highly trustworthy, if they don't indicate OP state the device is very likely not in OP state. It could be that initially your Python script is able to put the EL4004 into OP state, but it is falling back into SafeOP or PreOP state after a while. You may try to read the state a little later to confirm that.

To create a less complex example you could remove the other three EtherCAT terminals.

WeLikeIke, 2023-04-03:
Thank you very much for your response.

The actual problem was that writing to analog output takes a bit longer than 1 clock cycle to fully output 10V.

This meant that having in my code this loop for data exchange:
```python
while True:
    master.send_processdata()
    master.receive_processdata(10000)

    master.slaves[1].output = Aout_to_bytes(analog, 8, 4)

    if analog[0] >= 10.0:
        analog = [0] * 4
    else:
        analog[0] += 0.25
        analog[1] += 0.25
        analog[2] += 0.25
        analog[3] += 0.25

     # Timestepping, if absent the signal would be continuous
     # For example, for leaving a led on
     time.sleep(0.3)
```
I was interrupting the main data exchange for 0.3 seconds.

What I was not aware completely is that doing so would reset all digital and analog outputs to their default (off) state.

By using the same code but making the exchange continuous (removing the `time.sleep(0.3)` instruction) I was also able to correctly output between 0-10V and the lights on the EL4004 would behave as expected: turned on and NOT blinking.

Anyway, thanks again for the library and your time, I can close this now 👍

closed 2023-04-03 by WeLikeIke